prusa3d/PrusaSlicer #837: slic3r-prusa 1.39.1 reports that it is slic3r 1.39.0

hyperair, 2018-04-15:
### Version

1.39.1

### Operating system type + version

Ubuntu 17.10

### Behavior

SLIC3R_VERSION in libslic3r.h in https://codeload.github.com/prusa3d/Slic3r/tar.gz/version_1.39.1 is defined as `1.39.0`, which causes the wrong version to show up in the about dialog and generated gcode files.

stelgenhof, 2018-04-20:
I can confirm this as well. For the development branch version, it would be nice to indicate this is a development version.

bubnikv, 2018-05-10:
Thanks for the note. We have the version set at our build server from the last label with the "version_" prefix, but according to @hyperair I understand that the Debian build system does not work with git repositories, so this is difficult to him. I will try to be more careful to commit the version number into the source code when doing official releases.

> I can confirm this as well. For the development branch version, it would be nice to indicate this is a development version.

Silly me. The current master is marked incorrectly as 1.40.0. I have fixed that right now with https://github.com/prusa3d/Slic3r/commit/d620961c34ceb8ca747ab675042abb3abf8c341a

closed 2018-05-10 by bubnikv