prusa3d/PrusaSlicer #838: [BUG] Slic3r keeps warning about Detect Bridging perimeters if disabled

manuel-binpar, 2018-04-15:
### Version
1.39.1

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10

### Behavior
* _Describe the problem_
Slic3r keep warning about enabling "detect bridging perimeters" if disabled and "generate support material" is enabled.
* _Steps needed to reproduce the problem_
In "Print settings: Layer and perimeters", disable "detect bridging perimeters AND keep enabled "generate support material"
* _Expected Results_
The warning should be show once 
* _Actual Results_
The warning is shown every time that I enter to "print setting" to adjust any setting.

[image]

Additional note:

I'm using Slic3r with Repetier Host 2.0.5

Thanks!

bubnikv, 2018-05-10:
@FidelCapo Would you please verify this issue? Thanks.

FidelCapo, 2018-12-12:
This should be fixed in Slic3r PE 1.42 alpha, which is to be released soon. Closing.

closed 2018-12-12 by FidelCapo