prusa3d/PrusaSlicer #840: layer number in preview

maxdarkdog, 2018-04-17:
### Version
Slic3r-1.39.1-beta1-prusa3d-win64-full-201802221543

### Operating system type + version
Win7 Ultimate 67bit 

### Behavior
I think there is a big need to add a "layer number" indicator on the slice preview.

Why: 
I use a post processing Script to do multicolor prints as the one here (and many more i did with another slicer) 
[image]

The script is: 
{REPLACE "; layer" "M117 Layer"}
{REPLACE " Z = " " Z="}
{REPLACE "\nM117 Layer x, Z=" "\nM600\nM117 Layer x, Z="}

I need to know the layer number to replace fhe "x" so i can print in dual colors with a single extruder.... 

I really want to give a try on Slic3r by Prusa but this functionality is missing... 

_Is this a new feature request?_
YES 
#### STL/Config (.ZIP) where problem occurs
On the interface

Cheers ;)

bubnikv, 2018-04-17:
Actually there is a macro processing language in Slic3r PE
https://github.com/prusa3d/Slic3r/wiki/Slic3r-Prusa-Edition-Macro-Language

Not the most user friendly option, but with the Slic3r 1.38.5, you can now use the custom layer change G-code for that purpose. For example, the following line will trigger the color change action when changing to Z=5.2:

{if layer_z==5.2}M600{endif}

There may be an issue with numeric rounding, so it is safer to formulate the expression like

{if layer_z>5.19 and layer_z<=5.21}M600{endif}

bubnikv, 2018-05-10:
@enricoturri1966 Could we please fit the layer number indicator somewhere at the UI?

@maxdarkdog The problem may be, if you have multiple objects on the print bed, each sliced with a different layer height. And for the sequential print the situation will be yet different. What do you propose to show at the sliders?

bubnikv, 2018-08-02:
We show the layer index next to the layer Z height at the slides, certainly in Slic3r 1.40.0-alpha3. Closing.

closed 2018-08-02 by bubnikv