prusa3d/PrusaSlicer #841: underexrusion patern on wals

porlock, 2018-04-18:
### Version
Slic3r-1.39.1-prusa3d-win64-full-201803010855

### Operating system type + version
Windows 10 64bit

### Behavior
I get stange underexrusion  patern on wals where layer stars, i try almost everything

1) lowering retraction and detraction speed
2) Inceasing extruder stepper courrent
3) inceasing extruder spring tension

[problem.zip](https://github.com/prusa3d/Slic3r/files/1925874/problem.zip)

I also try other slicers and do not see this issue.

i record a movie when is clearly visible how my extruder works
https://youtu.be/6RJnntx5cNo

I attached all necesarry files, and pictures.

bubnikv, 2018-04-25:
I suppose you are printing the inner perimeters before outer perimeters, you also have the inner perimeter speed higher than the outer perimeter speed, and you have the cooling slow down enabled.

We are currently dealing with a similar issue. Normally the inner perimeters are printed faster than the outer perimeters, so you usually end up with a blob at the external perimeter, when the pressure from extruding the inner perimeter quickly releases. But with small objects and the cooling logic kicking in, the inner perimeters will be slowed down before the outer perimeters, so you will have a pressure deficit when extruding the outer perimeter.

This issue is similar to 
https://github.com/prusa3d/Slic3r/issues/536
https://github.com/prusa3d/Slic3r/issues/806
prusa3d/Prusa-Firmware#602

porlock, 2018-04-26:
is there any workaround to this issue ? should i print external perimeter first ?

bubnikv, 2018-04-26:
No, but you can print either multiple copies of the object at once, or you
may disable the cooling slow down, or you may lower the infill and inner
perimeter speeds close to the outer perimeter speed.

porlock, 2018-04-28:
i disable cooling slow down but it does not help, shout i do all things you metnion at the same time ?

porlock, 2018-04-28:
unfortunately version 1.39.2-beta helps nothing in my case :(
[image]

bubnikv, 2018-05-10:
Well, I have two hypotheses:

1) Insufficient cooling of the heat break. If you go faster or with higher temperature (sic!), it may get better.

2) You asked Slic3r to print a too thin wall, and Slic3r tries to fit two perimeters there. For example, you may get similar issues, if you let Slic3r print a 0.5mm thick wall with a 0.4mm extrusion width, and you have the "detect thin walls" disabled. Then Slic3r will try to squeeze two perimeters, nearly one over the other. In that case, either enable "detect thin walls", or make the wall thicker. In that case the Slic3r's hint will help. Also please note that the hint changes when you enable / disable "detect thin walls".

[image]

bubnikv, 2018-05-10:
Did it help?

porlock, 2018-06-02:
Unfortunetly nothing seams seams help :(
1) ady try faster speed and higher temp, but it make no difference :(
2) it occures also on thick walls 
[image]

closed 2026-09-29 by prusaslicer-bot[bot]