prusa3d/PrusaSlicer #842: Cut tool causes gcode export dialog to default to Slic3r executable directory

BManx2000, 2018-04-19:
### Version
1.39.1-prusa3d-win64

### Operating system type + version
Windows 10

### Behavior
When I use the cut tool on a model, the export gcode dialog opens on the location of the slic3r executable (C:\Program Files\Prusa3D\Slic3rPE). It should open on the location of the original STL file like it normally does.

_Is this a new feature request?_
No

tracernz, 2018-05-08:
It probably just defaults to the current working directory (this is a standard behaviour on all OSs). To check that, make a shortcut to the exe and set the working directory on it to somewhere else.

Area5142, 2018-05-08:
tracernz:
The behavior seems to be intentional.
If I import a STL model (drag and drop) and press "Export STL..." or "Export G-code..." the default directory is the same as the origin for the imported model (as expected).
But if I then press "Cut..."/"Split..." and then select "Export STL..." or "Export G-code..." the directory has changed to Prusa executable directory (unexpected).

I would prefer that the directory is not changed when using "Cut..." or "Split...", but remains the same as the imported model.

bubnikv, 2019-02-05:
This behavior cannot be reproduced with the Slic3r 1.43.0-alpha5. The piece of code responsible for the cut function and for the save / export has been reworked in 1.43.0-alphas, so it seems to be fixed.
Closing

closed 2019-02-05 by bubnikv