prusa3d/PrusaSlicer #843: Missing layers/gap in model causing print fails

dereulenspiegel, 2018-04-19:
### Version
1.39.1

### Operating system type + version
macOS 10.13.3

### Behavior
I am trying to slice a model from Thingiverse (https://www.thingiverse.com/thing:2787282). Unfortunately at a height of 23.2mm I get a gap in the model. At first I thought that this is probably a preview problem, but the actual print failed at exactly this part of the model. The model may be problematic so I tried repairing it in Meshmixer, without luck.
However Cura slices the model just fine and my print succeeded.
I hope that I am not reporting an already known issue, but I couldn't find something similar enough.

#### STL/Config (.ZIP) where problem occurs
[Link to the Slic3r config](https://www.dropbox.com/s/xc4871c1k6s4wom/slic3r_config.ini?dl=0)
[Link to problematic STL](https://www.thingiverse.com/thing:2787282)

brendjord, 2018-04-20:
i had similar problem with a fan shroud i tried to print. 
i fixed it with turning of "detect thin walls". maybe this will work for you to :)

bubnikv, 2018-04-20:
@FidelCapo Would you please check this?

dereulenspiegel, 2018-04-20:
@brendjord Thanks for the hint, but the problem persists wether I turn on or off the option "Detect thin walls". Although this time I didn't try to print it, but in the preview the layer ist still missing, so I guess it won't print anyway.

cerdeira, 2018-04-20:
In my opinion its a problem with the model it self.
To thin wall?
Simplify3d also shows problems when slicing it.

dereulenspiegel, 2018-04-20:
Ok, but for me as a user I want to have either one of two things:
* Slice the model as Cura does and create printable gcode
* Warn me about a problem with this model making it obvious for me that the generated gcode is probably not printable

Currently the only feedback I get with such models is my printer creating spaghetti :) 
So even if this model is problematic/broken I still see a bug or at least a chance to improve user experience here.

Itox001, 2018-04-21:
Are you slicing directly from slic3r's platter or using repetier? I get the same issue on 1.39.1 even with benchy when slicing from repetier host.

kfazz, 2018-04-21:
i'm hitting this same bug, using repetier host. the version of slic3r pe included with repetier (1.36.2-prusa3d-64) doesn't have this bug. 1.39.1 does. i was able to reproduce the failure on several different models. 

The attached zip contains the stls & scad i currently have this problem with.
[Prusa_i3_MK2_Clone_Extruder_for_8_12_18_mm_sensors_and_MK8_Hobbed_Pulley_1_Screw_better_cable_loop_edition.zip](https://github.com/prusa3d/Slic3r/files/1934787/Prusa_i3_MK2_Clone_Extruder_for_8_12_18_mm_sensors_and_MK8_Hobbed_Pulley_1_Screw_better_cable_loop_edition.zip)

LarsTT, 2018-04-21:
Just joining in here. I'm having the same problem when trying to print these:
https://www.thingiverse.com/thing:2410748

Slic3r Prusa Edition, Version 1.39.1-prusa3d-win64
I've checked the generated G-code in both Repitier Host and S3D.
However S3D slices them without problem, both slicers are set with the same manual extrusion width.

[image]
[image]

kfazz, 2018-04-21:
just confirming that the missing layers seen in the preview affect the gcode as well. I tested the previous release, version 1.38.5-prusa3d-64. It does not have the issue, and I was able to correctly slice the stls i was having a problem with.  
Also, i was able to make a correct file fail by changing the $fn value in the scad, specifically, changing "$fn= n" to "$fn = 360" in polyholes.scad. That caused the x-idler (uses polyholes) model to fail. Maybe model complexity has something to do with it?

dereulenspiegel, 2018-04-22:
I am using Slic3r Platter.

vernieg, 2018-04-22:
#813  is a similar issue.

bubnikv, 2018-05-10:
@FidelCapo Would you please test this G-code with our current 1.40-alpha? Would you please add the models to our database and if the 1.40-alpha processes this model correctly, would you please close this issue? Thanks.

FidelCapo, 2018-05-15:
Tested and fixed in Slic3rPE-1.40.0-alpha. Closing

closed 2018-05-15 by FidelCapo