prusa3d/PrusaSlicer #844: ChangeRequest - move settings to where they belong

DanyboyInSwitzerland, 2018-04-20:
### Version
_1.39.1-prusa3d-win64_
### Operating system type + version
Win10Pro, Version 1709, OSbuild 16299.371

### Behavior
Feature/Change request
Printing-speeds are heavily depending on the material. However, the speed-settings are within the print-settings, where procedural aspects and geometries are defined, like how many perimeters, what type of infill, etc. All Settings, that are not depending on the type of material getting used.
This means that for every type of material, one can save a set of settings.
After a fresh install, the user has the four settings Ultradetail, Detail, Optimal and Fast.
But already when using Flexfill and PLA, the number goes from four to eight settings, since Flexfill must be printed way slower than PLA.
If the two parts "Speed for print moves" and "Autospeed" are moved into the tab "Filament Settings", then the speeds could be saved in relation to the material used - which is what they're depending upon, the figures are then where they truly belong.
If different speeds are needed for the different layer-heights, "Speed for print moves" and "Autospeed" could be changed to a percentage, which in turn would reference to the related setting defined in the "Filament Setting". The speeds under filament-setting would be the nominal speed, if you want to go faster in profile "0.20 FAST MK3", you can define 120% there, if you want to go slower in profile "ultradetail", you can put 80% there. And since you always want to print slower when using small layer-height, these 80% apply to all materials...
The whole process of speed-settings would get more modular (since there are now two places that influence speed), but the number of profiles would be drastically reduced, and the figures entered have a clear purpose. 

#### STL/Config (.ZIP) where problem occurs
All.

rongith, 2018-04-25:
It is apparently already under study: https://github.com/prusa3d/Slic3r/issues/480
And Alexrj implemented an override feature for 1.3.0 milestone: alexrj#3643 in original Slic3r.

DanyboyInSwitzerland, 2018-04-26:
Hello rongith,
Oh, so obviously despite browsing and looking for I did not see it... My bad...
Thanks for the reply!
And please keep up the work!The only problem I really have are the supports - otherwise it's a great piece of software! :)
Cheers,Daniel

  belong (#844)

bubnikv, 2018-05-10:
There is no perfect split of the parameters into the print / filament / printer preset.

We are certainly looking into some form of overriding print / printer parameters by the filament presets and similar tricks, but it will not be part of the upcoming 1.40 yet.

bubnikv, 2019-08-08:
retract overrides at the filament level will be part of the upcoming PrusaSlicer 2.1.0-alpha1. Closing this issue. Please open another issue if you feel you need to override another parameters. We expect that there is no need to override any other Printer or Extruder parameter, but there may be a need to override some of the Print parameters.

closed 2019-08-08 by bubnikv