prusa3d/PrusaSlicer #846: Sliced Info and Info exceed view port (partly invisible) on Plater UI

stelgenhof, 2018-04-21:
### Version
`git rev-parse HEAD`: 987ef3730a9ef43244022013829e8aeb2cc501e5

### Operating system type + version
Arch Linux (Kernel: 4.15.15-1-ARCH)

### Behavior
The Sliced Info and Info information exceeds the view port size on Plater UI. The view port is not scrollable so part of the information is invisible (See screenshot).

[image]

YuSanka, 2018-05-13:
Thanks a lot for your report. Fixed bug is merged in the current master, next public release is expected in several days.

stelgenhof, 2018-05-14:
Thanks! I can see that the info window now contains scrollbars allowing to view all the information. However I noticed two additional things:

1. Upon start, the settings area on the right is rather small making the text looking kind of compressed/overlap. This does get corrected once you load a model, because the main 3D window size gets reduced somewhat.

[image]

2. The area on the right displaying the different objects seem to be fixed in size causing the information (sliced results, time etc.) to move all the way to the bottom of the viewport. Of course you can use the scrollbars to scroll down to that information, but previously the information was just visible without taking an extra action of scrolling

[image]

Not major issues, just wanted to let you know :)

bubnikv, 2019-05-21:
Is PrusaSlicer-2.0.0 work correctly for you? Can you close this issue, please?

stelgenhof, 2019-05-21:
@bubnikv Let me compile v2.0 on my machine and let you know if it has been fixed.

stelgenhof, 2019-05-21:
[image]

@bubnikv Unfortunately my compilation failed. Apparently an image seems to be missing.

stelgenhof, 2019-05-21:
@bubnikv I can confirm the issue has been fixed in v2.0 of PrusaSlicer on Arch Linux. Shall I create another issue for the earlier mentioned bitmap compile error?

lukasmatena, 2019-05-22:
@stelgenhof Hi. Are you sure you're building what you think you are? I just searched the current source code and it does not contain a single occurence of "sys_unlock" string. Also, I just verified that the title of the message box correctly states "PrusaSlicer". Yours doesn't.

stelgenhof, 2019-05-22:
@lukasmatena Yes, I do :) It compiles without any issue. BTW, "sys_unlock" exists in the source, see commit 1c5970d0a2ea0a4fde1af36502ea93b577208504.

Anyway the issue was that I had a shortcut to the compiled binary and that was apparently renamed to 'prusa-slicer'. Executing that binary was the key :)

Thanks! (Closing the issue).

closed 2019-05-22 by stelgenhof

bubnikv, 2019-05-22:
cool, thanks for testing.

lukasmatena, 2019-05-22:
@stelgenhof Ok. But I suggest looking at https://github.com/prusa3d/PrusaSlicer/commit/56256519cc122d440d851c8d3a41d11ba904b4b1 too then. The world is changing.

stelgenhof, 2019-05-22:
@stelgenhof Thanks! I must have missed that. I've already corrected my shortcut to the appropriate binary. All works fine now...