prusa3d/PrusaSlicer #849: Skirt / Brim generation is not respecting table operating size

karek314, 2018-04-22:
### Version
1.39.1-prusa3d

### Operating system type + version
macOS 10.13.4 (17E199)

### Behavior
Skirt / Brim generation is not respecting table operating size

If object is placed very near to table end, brim and/or skrit is generated outside table, hence gcode for printer is also trying to print outside operational dimensions.

3Delights, 2018-05-08:
I have the same issue with version 1.39.2 on Windows (10).

I also noticed that the support material extends beyond the edges of the bed also.

bubnikv, 2018-05-10:
You are completely right, there is currently no check for the supports or brim/skirt against the working space.

bubnikv, 2018-08-02:
We have a check in the current Slic3r-1.41.0-alpha3 in the G-code preview window. Does it satisfy your needs?

3Delights, 2018-08-02:
All it does is put up a warning telling you that it is off the print bed?  You can already see that just by looking at the preview!

[image]

It needs to actually stop the brim (and supports) being generated or only create it(them) on the bed.

Stratagm, 2023-11-10:
Over 5 years on, and still no love 💔

IMB11, 2025-01-22:
Its 2025 and we still have this issue...

3Delights, 2025-03-23:
People are complaining about this problem on the forum!  It's been 7/8 years since first reported, it should be fixed by now!  If the code can detect that the print is outside the bed then why can't it do so while generating supports, brims, etc.

On a related note, support blockers don't block these errant supports either!

jaredglaser, 2025-06-11:
This is the feature forcing me to use Cura still... Prusaslicer works so much better otherwise. Even with manual painting you can't define where on the buildplate the organic supports should originate so you just can't use it if the item is too close to the build plate border.

I have some items I print that have large open spaces in the middle where the supports could start from ( and do in Cura ), but Prusaslicer will always put them on the outside.

murk-sy, 2025-06-28:
The issue [where I posted this](https://github.com/prusa3d/PrusaSlicer/issues/12677#issuecomment-2092923454) got closed as duplicate so I'll put it here for future reference.

-----

### Workaround for SUPPORTS outside build plate (not applicable to skirt/brim)

Original issue image:
 [image]

Since support blockers don't do much, what you can do is place very thin (0.05 mm) parts that cannot be printed right next to the part. Edit: They need to be merged with the main object, visible in the downloadable project.

[image]

[image]

[Support-with-workaround.zip](https://github.com/prusa3d/PrusaSlicer/files/15199958/Support-with-workaround.zip)

j28CLY, 2025-07-08:
@murk-sy Thank you for that workaround. 
I'll add that it may be necessary to add multiple layers of these thin walls. 
I'm working on something mostly simple and rectangular, with straight side walls, and PrusaSlicer insisted on adding supports on the outside of the part for some reason, serving only to support some air.
I tried blocking them with this workaround, but PS adapted and maneuvered new supports to start growing outside of the walls. With three layers of walls in place, spaced about 4mm apart, it finally gave up on trying to make these overzealous supports.

Yes, organic supports really need to stop doing things like venturing outside of the build-volume, or into support-blocker volumes.

NobodysNightmare, 2026-01-05:
Thanks for the workaround, maybe I can solve my similar problem this way as well. Right now I just get the support base cut in half:

[image]

But maybe that will lead me somewhere.

P.S.: If you don't want to add the thing to the original file, the workaround also seems to work with a cube added in PrusaSlicer.

Tupson444, 2026-09-12:
not fixed in 3.0.0

IMB11, 2026-09-13:
> This issues has been labeled legacy. It means that it has been created before the release of PrusaSlicer 3.0.0 alpha version. In an attempt to clean up the GitHub issues tracker any legacy issue [...]

Ridiculous - an open issue for the past 8 years and it's still not fixed.