prusa3d/PrusaSlicer #850: Slic3r does not limit height of objects

karek314, 2018-04-22:
### Version
1.39.1-prusa3d

### Operating system type + version
macOS 10.13.4 (17E199)

### Behavior
Slic3r does not limit height of objects in respect of printer config.

enricoturri1966, 2018-04-23:
Out of print volume detection (taking in account also the height) has been introduced starting with: https://github.com/prusa3d/Slic3r/commit/bdd2d725c83fca0d3a84f19cf9ae25ce2525cbb9

karek314, 2018-04-25:
For which version of release is it planned to include https://github.com/prusa3d/Slic3r/commit/bdd2d725c83fca0d3a84f19cf9ae25ce2525cbb9

bubnikv, 2018-04-25:
> For which version of release is it planned to include bdd2d72

Slic3r PE 1.40. We are reaching an internal alpha as of now.

closed 2018-04-25 by karek314