prusa3d/PrusaSlicer #853: Sliced model has big gap in it

ccfoo242, 2018-04-26:
### Version
1.39.1-prusa3d-win64

### Operating system type + version
Windows 10

### Behavior
Slice MainBody.stl from [https://www.thingiverse.com/thing:2844119](url)
Big gap in part of it. Tried with Cura 3.3 and it slices correctly.
If you rotate on its side it slices correctly.

#### STL/Config (.ZIP) where problem occurs
[image]

[config.zip](https://github.com/prusa3d/Slic3r/files/1949483/config.zip)
[MainBody.zip](https://github.com/prusa3d/Slic3r/files/1949485/MainBody.zip)

redangel1984, 2018-04-26:
I observed the same issue but only with version 1.39.1 here are two of
model is:
https://www.thingiverse.com/thing:847941

and used the same settings on both versin with different result(see below):
[Slic3r_PE_config_0.3mm_ini.zip](https://github.com/prusa3d/Slic3r/files/1950073/Slic3r_PE_config_0.3mm_ini.zip)

1.39.1
[image]

1.38.5
[image]

bubnikv, 2018-04-26:
@FidelCapo Please retest with 1.40-alpha and put the model into our set of test objects. Thanks.

redangel1984, 2018-04-26:
I thought the issue might be the key ring, but after splitting the model, the layers are still missing:
[image]

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

mvasilakis, 2018-04-26:
@bubnikv is 1.40-alpha pre-compiled?

bubnikv, 2018-04-26:
> 1.40-alpha pre-compiled?

We don't really want to share the alphas, but there will likely be a public
beta released in a week or two.

mvasilakis, 2018-04-26:
@bubnikv I figured as much but since you mentioned it above I figured I'd ask. Looking forward to the beta 😃 

FWIW I had a similar break in a lithophane made from 3dp.rocks and mesh corrected in mesh mixer. It displayed fine on screen unlike OP's but the g-code was split in 2 parts and floating in thin air about 20 mm behind the model at the correct z height. 

I unchecked Ensure vertical shell thickness, avoid crossing perimeters, and detect thin walls and the resulting slice was correct. Not sure if OP has any of those checked but it might be worth a peek.

ccfoo242, 2018-04-26:
@mvasilakis  I did try detect thin walls, which helped with this problem on a different model, but it didn't change this one. Haven't tried the other options.

mvasilakis, 2018-04-26:
@ccfoo242 I had to remove those options to fix the model in my case. If they're not selected then one of those is not the issue. Possibly just a similar problem caused by a different reason.

bubnikv, 2018-05-10:
@FidelCapo Please retest with 1.40-alpha and put the model into our set of test objects. If the alpha 1.40 processes these models, please close this issue. Thanks.

FidelCapo, 2018-05-15:
Tested and fixed in Slic3rPE-1.40.0-alpha. Closing

closed 2018-05-16 by bubnikv