prusa3d/PrusaSlicer #855: Incorrect slicing of a fractal-like model

vojtechkral, 2018-04-27:
### Version
Current 1.40 alpha, 1.39.2

### Operating system type + version
Current Arch Linux

### Behavior

Gaps are being enclosed from above by other layers.

[image]

#### STL/Config (.ZIP) where problem occurs

The STL is generated by OpenSCAD 2015.03-3 using the following code:

```
module rec4(size, step) {
    s2 = size / 2;
    s4 = size / 4;
    translate([0, -s2, -s2]) cube([s2, s2, 3*s4]);
    translate([-s2, -s2, -s2]) cube([s2, s2, s2]);
    translate([-s2, 0, -s2]) cube([s2, s2, s4]);
    

    snext = s2;
    if (step > 0) {
        translate([s2/2, -s2/2, s2]) rotate([0, 0, 90]) rec4(snext, step - 1);
        translate([-s2/2, -s2/2, s4]) rotate([0, 0, 0]) rec4(snext, step - 1);
        translate([-s2/2, s2/2, 0]) rotate([0, 0, -90]) rec4(snext, step - 1);
        translate([s2/2, s2/2, -s4]) rotate([0, 0, 180]) rec4(snext, step - 1);
    }
}

scale(5) rec4(10, 4);
```

vojtechkral, 2018-05-01:
@FidelCapo STL file:
[recur-bug-2.stl.zip](https://github.com/prusa3d/Slic3r/files/1963436/recur-bug-2.stl.zip)
(pretty much any configuration)

bubnikv, 2018-05-17:
Which layer do you think is incorrect? It seems to me that the cavities are part of the model.

vojtechkral, 2018-05-18:
> It seems to me that the cavities are part of the model.

Shouldn't be. I've verified with meshlab and cura, you should be able to see the "ground" from above.
The problem is on layers 40 through 61 when slicing 0.2mm layer height & default settings.

bubnikv, 2018-05-18:
There are indeed some overlapping triangles visible even in Slic3r with the new feature of visualizing the inverted faces with dark colors.

[image]

Rendering the slice nr. 39 shows a mess of contours. The blue is correct, the rest are pieces of correct and overlapping geometry.

[image]

Implementing a robust algorithm for slicing such a messed up model is not easy. We may keep this issue open for reference, but to fix it we need to collect more examples to derive a reasonable set of rules how to handle these issues.

closed 2026-09-29 by prusaslicer-bot[bot]