prusa3d/PrusaSlicer #857: Visual Studio XS Project incomplete

bjku96, 2018-05-01:
### Version

Slic3r Prusa Edition 1.39.1

### Operating system type + version
Windows 7 64bit

### Behavior

Hello,
I'm trying to build slic3r with Visual Studio 2013. I did everything described in this tutorial:
 https://github.com/prusa3d/Slic3r/blob/master/doc/How%20to%20build%20-%20Windows.md 
When I use Nmake to compile it, everything works fine and slic3r can be launched via Perl. 

The Problem occurs when I try building the project files fo VS. 
Cmake (version 3.11.0) creates the necessary files in my build directory with this command:
**cmake .. -G "Visual Studio 12 Win64" -DCMAKE_CONFIGURATION_TYPES=RelWithDebInfo**

But when I try to build slic3r in VS with F5 I get this output: 	

```
12>------ Build started: Project: XS, Configuration: RelWithDebInfo x64 ------
12>  Generating typemap
12>  Building Custom Rule C:/local/Slic3r/xs/CMakeLists.txt
12>  CMake does not need to re-run because C:/local/Slic3r/build/xs/CMakeFiles/generate.stamp is up-to-date.
12>  Generating XS.cpp
12>  Der Befehl "xsubpp" ist entweder falsch geschrieben oder
12>  konnte nicht gefunden werden.
12>C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V120\Microsoft.CppCommon.targets(170,5): error MSB6006: "cmd.exe" exited with code 9009.
========== Build: 11 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========

```
I think the problem is that the XS-Project is incomplete. When I tried to open XS.cpp I got this error:
[image]

I don't know if there are more files that are missing. 

With the 32 bit variant
**cmake .. -G "Visual Studio 12" -DCMAKE_CONFIGURATION_TYPES=RelWithDebInfo**
the same error occurs.

vojtechkral, 2018-05-02:
@bjku96 Which Perl bundle are you using? Do you have either `wperl64`, `wperl64d`, or `wperl32` directory in your `C:\`?

I think the bug is that in the MSVC props file I only added an entry for `wperl64d` and forgot the others. Will fix.

vojtechkral, 2018-05-02:
@bjku96 Hopefully the above commits should fix the problem. Would you be so kind as to re-test with the two above commits (they're in `master`)? You'll have to re-generate the Visual Studio files.

bjku96, 2018-05-02:
**@vojtechkral**  So, to make sure that there is nothing else wrong with my slic3r-files I cloned them again from git, executed build.pl  and the cmake command. (This time with wperl64 )
The XS-Project is still incomplete. (XS.cpp is missing). 
I already used wperl64**d**  before and had the same problem. 
I dont know if it has something to do with this but when Build.PL tries to install all dependencies for the gui , installing wx-0.9932 always fails and the build-log says: 

```
Microsoft (R) Program Maintenance Utility Version 12.00.21005.1
Copyright (C) Microsoft Corporation.  All rights reserved.

	"C:\wperl64\bin\perl.exe" script/make_v_cback.pl > cpp/v_cback_def.h
Can't open perl script "script": Permission denied
NMAKE : fatal error U1077: '"C:\wperl64\bin\perl.exe"' : return code '0xd'
Stop.
```
Had a similar problem when I tried to install PAR::Packer. 
I don't know why perl has no permission to open this file.

bubnikv, 2018-05-10:
Don't you use windows reparse points aka sym links in your paths where you have the Slic3r or any of the depending components installed? Don't you use any non-latin1 characters in some of the paths?

bubnikv, 2018-08-02:
No response, we were not able to reproduce the issue. Closing.

closed 2018-08-02 by bubnikv