prusa3d/PrusaSlicer #858: Printing time calculation completely wrong

ir-fuel, 2018-05-02:
### Version

1.39.1-beta1-prusa3d

### Operating system type + version
macOS 10.12.6

### Behavior

Running a Prusa i3 mk2s MMU print, it tells me the estimated print time is 8h 31m, but my printer does it in approx 3h

#### STL/Config (.ZIP) where problem occurs
[config.ini.zip](https://github.com/prusa3d/Slic3r/files/1966871/config.ini.zip)
[image]
[test (1).gcode.zip](https://github.com/prusa3d/Slic3r/files/1966872/test.1.gcode.zip)
[test.stl.zip](https://github.com/prusa3d/Slic3r/files/1966874/test.stl.zip)

screwyluie, 2018-05-07:
you're not the only one, mine is completely wrong as well. It'll tell me something crazy like 21 hours for a 2 hour print. It's never been right, not even close.

bubnikv, 2018-05-07:
@enricoturri1966 looked into your test. Please set the firmware flavor to Marlin, it will work correctly then.

screwyluie, 2018-05-07:
interesting. That got it a hell of a lot closer. Used a known print that was 3h12m now slic3r says it's 2h40m. A second known print that's 2h24m slic3r says 2h7m.

thanks @bubnikv

bubnikv, 2018-05-17:
@screwyluie You should use Slic3r's MK2 profile for the MK3. No wonder that the estimates are not quite on spot. 

I believe we have solved this issue. Closing.

closed 2018-05-17 by bubnikv

ir-fuel, 2018-05-17:
@bubnikv  I don't really understand. 

I opened this topic with an mk2s MMU time error. What has this got to do with "You should use Slic3r's MK2 profile for the MK3." ?

Seems to be you only replied to the 2nd poster in this thread's problem?

bubnikv, 2018-05-17:
Yes, I answered @screwyluie 's issue.

Anyway, tested with Slic3r 1.40.0-alpha. Your MMU print, after I switched the firmware type from Reprap to Marlin, changed to the expected 2h50m. So I rightfully closed this issue.

ir-fuel, 2018-05-17:
So is it a bug in the default printer profile that Prusa provides? I have never touched that setting.

bubnikv, 2018-05-17:
I hope our profiles are correct, more likely you did not update the profiles when installing Slic3r 1.39.

Luckily starting with the Slic3r PE 1.40 the profiles will be updated from the internet automatically.

ir-fuel, 2018-05-17:
These are the latest profiles as found on Github, "old style" (so pre 1.40). Check that line.

https://github.com/prusa3d/Slic3r-settings/blob/3611992e48ba3dd36c77d0234604bbb438678882/old/Slic3r%20settings%20MK2S%20MK2MM%20and%20MK3/printer/Original%20Prusa%20i3%20MK2%20MultiMaterial.ini#L9

bubnikv, 2018-05-17:
> 3611992e48ba3dd36c77d0234604bbb438678882/old/Slic3r%
> 20settings%20MK2S%20MK2MM%20and%20MK3/printer/Original%20Prusa%20i3%20MK2%
> 20MultiMaterial.ini#L9

Thanks, I sent a note to Jindrich, who takes care of the profiles.

Luckily the new hierarchical form of profiles will make such errors less
probable.

https://github.com/prusa3d/Slic3r-settings/blob/3611992e48ba3dd36c77d0234604bbb438678882/live/PrusaResearch/0.1.2.ini

You see, there is just a single gcode_flavor line applied to all profiles
of Prusa3D.

ir-fuel, 2018-05-17:
Yes, I saw it too. A lot less duplication. Good move!