prusa3d/PrusaSlicer #860: Missing layer

BirkBinnard, 2018-05-02:
### Version
1.39.2 Beta

### Operating system type + version
Windows 10/64

### Behavior
Slicing resulted in what looks like 1 missing layer and 1 over-thick layer. This object is the egg from https://www.thingiverse.com/thing:2887652 scaled up to 200% size

[image]

#### STL/Config (.ZIP) where problem occurs

[jet_fighter_egg.zip](https://github.com/prusa3d/Slic3r/files/1968029/jet_fighter_egg.zip)

stelgenhof, 2018-05-06:
@BirkBinnard Sliced this one myself as well and don't have this issue. I'm using v1.40 (dev version) running on ArchLinux. 
Can you maybe share your parameters (Print Settings, Filament settings)?

BirkBinnard, 2018-05-06:
Sure - here's my config.ini file (renamed config.txt). I have a delta style printer with a round printbed. My guess is the issue has to do with layer height - but that's just a guess. Filament is 1.75 mm dia.

[config-bb.txt](https://github.com/prusa3d/Slic3r/files/1977342/config-bb.txt)

I'd be happy to re-do the test with the same version you have :)

stelgenhof, 2018-05-06:
Thanks! Just used your configuration, scaled it to 200% however the results still look good on my side.
I will have a look in v1.39 as well.

bubnikv, 2018-05-06:
we plan to release a 1.40 alpha very soon, it should fix these issues.

FidelCapo, 2018-05-15:
Tested and fixed in Slic3rPE-1.40.0-alpha. Closing

closed 2018-05-15 by FidelCapo