prusa3d/PrusaSlicer #861: Small bits of support are generated where there is nothing to support.

ManuelMcLure, 2018-05-03:
### Version
1.39.1-prusa3d-win64

### Operating system type + version
Windows 10 64 bit

### Behavior
* _Describe the problem_
Small bits of support material get generated where there is nothing to support
* _Steps needed to reproduce the problem_
Slice the attached STL with the attached config.
* _Expected Results_
Support only generated where it's supposed to be (under the hexagonal cutouts below the pillars).
* _Actual Results_
Small bits of support are generated in the large clear area next to the pillars - see circled area* _Screenshots from __*Slic3r*__ preview are preferred_
[image]

_Is this a new feature request?_
No.

#### STL/Config (.ZIP) where problem occurs
[STL_and_Config.zip](https://github.com/prusa3d/Slic3r/files/1969541/STL_and_Config.zip)

bubnikv, 2018-05-17:
That's the 1st layer of the support leaking from that supported hole. That is really just a cosmetic, harmless issue, though we may fix it one day.

ManuelMcLure, 2018-05-17:
I figured it might be something like that. It's obviously nothing more than a small annoyance but I wanted to ensure that there was awareness of the issue.

bubnikv, 2021-02-24:
Fixed with 539e80e890abcdd7c19233339f577e1833f203d4. Duplicate of #2245

    When expanding the 1st layer support layer, do it by small steps while
    trimming with object to avoid supports to leak through object walls.

closed 2021-02-24 by bubnikv