prusa3d/PrusaSlicer #862: Contents of the GCode fields are not visible

stelgenhof, 2018-05-03:
### Version
1.40.0
git rev-parse HEAD: 20d9bda87e8a8e83a6a9c905120156180d5e935d

### Operating system type + version
Arch Linux 4.16.5-1-ARCH

### Behavior
The contents of the gcode fields in the "Printer" tab are not visible. I have created a printer preset based on the system MK2S preset. Even the start/end gcode fields of the system preset are not visible.

If I try typing in on of these fields, the program registers that I made a change however nothing is visible. When I save my preset, then I can see with a text editor that the gcode field has changed actually.

It seems that the GUI is not working properly (maybe related to color scheme / font colors).

[image]

YuSanka, 2018-05-03:
@stelgenhof , Thanks again :) This bug is fixed https://github.com/prusa3d/Slic3r/commit/af4e0308ae63149a5bc7bcf69b67d0967c7900b2

stelgenhof, 2018-05-03:
@YuSanka Thanks for the quick fix! Confirm it has been resolved.

closed 2018-05-03 by stelgenhof