prusa3d/PrusaSlicer #864: 3D and Preview Windows not rendering correctly

Jonny-Cache, 2018-05-04:
### Version
1.39.0

### Operating system type + version
Ubuntu 18.04

### Behavior
3D and Preview windows only rending in the bottom left quadrant of the screen.
Likely a HiDPi vs Linux issue (running on a 4k monitor, Slic3r runs fine on the same screen in Windows 10).

[image]

bubnikv, 2018-05-17:
@vojtechkral @lukasmatena are you guys able to reproduce it?

We know that wxWidgets 3.1 improved upon the HiDPI support, but we cannot easily upgrade to wxWidgets 3.1 due to the Perl binding, which is not compatible with it. We may try, but we may invest our time better by trying to get rid of the Perl as quickly as possible.

vojtechkral, 2018-05-18:
I'm unable to reproduce the problem with either 1.39.2 or current `master` :-/ (But then again my screen is only 2560x1440.)

vojtechkral, 2018-05-18:
I'm not sure how to do that (or if there is such a thing). I tried changing the scaling factor on that screen to various settings (75%, 150%, 200%) but Slic3r always rendered normally.

lukasmatena, 2018-05-18:
I didn't manage to reproduce it either (Ubuntu 16.04), changing scaling-factor did rescale the user interface, but Slic3r looked fine (but I don't know if its enough to provoke the problem) Apparently this is not the first time someone had this problem: https://github.com/slic3r/Slic3r/issues/3362. The thread makes an impression the problem appeared after linking against wxWidgets with GTK3 support, it might be interesting to know whether @TheCodeSharman managed to get rid of the problem and how.

TheCodeSharman, 2018-05-18:
Hi, yes a while back I took a look at this. 

Unfortunately its all about the lack of maintenance for wxPerl. What appears to be happening is that Gtk3 is automatically scaling the windows to take into account the higher dpi settings. See here: https://wiki.gnome.org/HowDoI/HiDpi

Since OpenGL doesn't know about this scaling factor what needs to be done is to read it from GTK and multiply the provided width/height in GetSizeWH to compensate on HiDPI.

wxWidgets 3.1 exposes this as GetContentScaleFactor() in interface/wx/window.h

So AFACIT the code code change is in Slic3r::GUI::3dScene in the Resize() method to multiply the coordinates of the size rectangle by the result of $self->GetContentScaleFactor()

There also exists $self->ToDIP and $self->FromDIP (see http://docs.wxwidgets.org/3.1/classwx_window.html#aa934a31552633c076a87bca1100b0dc5)

But given that GTK already uses device independent pixels I think these won't give us the physical pixels that OpenGL is expecting.

Unfortunately these APIs don't work the GTK version  of wxWidgets 3.0 so we need 3.1. This is where it begins to get messy.

Seems like Slic3r already has a solution for this:
https://github.com/slic3r/Slic3r/wiki/Building-wxWidgets-3.1.0

But this is failing on my machine at the moment. In the end, I started running Slic3r on a machine that doesn't have a HiDPI screen so I gave up at this point.. 

BTW: The reason I tried to compile a Gtk 3.0 version of Slic3r was that the AppImage version was very difficult to use because of the fonts not scaling very well. Since then I've been using Ubuntu 18.04 based Pop!_OS (https://system76.com/pop) and things are a lot better with the AppImage install. 

I'm not sure what's changed, possibly some font scaling settings for GTK 2 ?

There are still a few issues in terms of needing to horizontally scroll panel views and the "Export STL" and "Slice Now" buttons are mostly unreadable but still functional.

Jonny-Cache, 2018-05-24:
Sorry for the delay! I tried the 1.39.2 beta appimage version and it's behaving better now.
The dark desktop theme is not being respected and the Icons for the buttons are much smaller, but the original issue with the 3D window is now resolved and working well.

Please feel free to close out this issue as the original problem with the 3D render pane is clearly resolved in the latest beta. :)

[image]

bubnikv, 2018-06-04:
@Jonny-Cache Would you please test the 1.40.0-beta appimage? If that works for you, I will be more than happy to close this issue.

So it seems to me, that the issue is when running the wxWidgets 3.0 against GTK3? I also know @YuSanka did quite a lot of improvements of the GUI layout.

SijmenSchoon, 2018-06-20:
For what it's worth, it works fine on V1.40.0:

[image]

OS: Arch Linux
WM: GNOME 3 on Wayland
Display: 3840x2160 with 200% scaling

tracernz, 2018-06-20:
Also fine here on Arch with Gnome 3.28 on Wayland. :+1: I think Ubuntu is still on Xorg by default though (they backed out of their plan to switch to Wayland quite late in their release cycle)?

bubnikv, 2018-06-28:
So it seems to work now for everyone involved in this discussion. Happily closing.

closed 2018-06-28 by bubnikv

ArchangeGabriel, 2019-03-25:
I’m having the same issue when building the latest 1.42-beta1 under ArchLinux. I’ve compiled against the repo WxGTK3, so if I understand things correctly, I should either move to WxGTK2 or see if we can upgrade to WxGTK3 3.1. Correct? I will investigate those options ASAP, though I’m a bit busy lastly…

tracernz, 2019-03-25:
The AppImages on the releases page run fine on Arch, with no HiDPI issues. I've run previous versions under Wayland, and now I've gone back to Xorg for FreeSync, both no issues.

bubnikv, 2019-03-25:
We are building on the build server against GTK2 and wxWidgets 3.1.
Our developers may use a different setup though.

@Lukas Matena <lukasmatena@seznam.cz> @Vojtech Bubnik
<bubnikv@gmail.com> @Mészáros
Tamás <meszaros.q@gmail.com>
What combination of GTK and wxWidget versions are you guys using for
development?

bubnikv, 2019-03-25:
@vojtechkral

ArchangeGabriel, 2019-03-25:
@tracernz I’m not talking about AppImage here, but an incoming Arch package in [community].

We are looking at upgrading wxWidgets to 3.1, I will test again GTK3 and if it still doesn’t work or we can’t manage to move to wxWidgets 3.1 because of incompatibilities with other software, I’ll try GTK2.

vojtechkral, 2019-03-25:
FWIW, I'm trying out wx3.1 + GTK3 on my Arch.

ArchangeGabriel, 2019-03-25:
I did not have time to look at it yet, but apparently we already faced some issues while trying to rebuild other packages for wxWidgets 3.1… So I will likely have to go the GTK2 path, will see and keep you updated.

vojtechkral, 2019-03-25:
Well the main issue is that wx 3.1 is not in repos, right? (AUR only)

ArchangeGabriel, 2019-03-25:
No, the main issue is that we started a rebuild against wx 3.1 ((https://www.archlinux.org/todo/wxgtk-31-rebuild/) but some software fail to build (amule and python2-wxpython3 at least). There might be a way to have both wx version though, but no time on my side to work on this right now.

vojtechkral, 2019-03-25:
@ArchangeGabriel Ok, thanks for info

nordurljosahvida, 2020-12-14:
@bubnikv I'm having the same issue here on Ubuntu 20 with GTK3 appimage [as first image on top, the canvas is 1/4 the area in bottom left corner]. Non-GTK3 appimage works fine. I have my screen at 200%. Mentioning you as I don't think this would be seen having already been closed otherwise. Thanks!