prusa3d/PrusaSlicer #865: Bug when slicing, creating an infill layer with 100% infill at random height

Snorvarg, 2018-05-04:
### Version
1.39.1-prusa3d-win64

### Operating system type + version
Windows 10 Pro, "latest" version. (Where do one find the version of windows nowadays..) 

### Behavior
* The slic3r creates an infill layer randomly, destroying any cavities with a sudden layer of infill.
* Choose 0.20mm NORMAL, then select 100% fill density, switch to rectilinear fill pattern. Slice.
* A cavity inside a cube.
* At some random height, the cavity gets filled with a infill layer. See screenshot.

NOTE: This has happened with another model as well. The problem disappears if I change to 0.10mm DETAIL, or any other preset. (with 100% fill density)

NOTE: I noticed the fill density is NOT part of the problem; 20% fill density also creates this problem.

Screen dump showing the unwanted layer:
[image]

The very next layer is correct:
[image]

#### STL/Config (.ZIP) where problem occurs
[embedded-skull-die-for-transparent-3d-printing-by-joealarson.zip](https://github.com/prusa3d/Slic3r/files/1974374/embedded-skull-die-for-transparent-3d-printing-by-joealarson.zip)

[config.txt](https://github.com/prusa3d/Slic3r/files/1974388/config.txt)

bubnikv, 2018-05-16:
I just tested on Windows with the 1.40.0-alpha1, and it seems to be fixed. Would you please confirm?

https://github.com/prusa3d/Slic3r/releases/tag/untagged-8fc8ca127d0924fe5504

DrLex0, 2018-05-16:
This looks a lot like one of the two problems I've seen in #895.

Dainse, 2018-05-18:
I have tried the new version that you have suggested (1.40.0 alpha1) with the same model and the slicer is perfect, it seems that the error has been corrected in this version

bubnikv, 2018-05-18:
thanks for the report. I am still fighting with the 2nd issue of #895, which sounds too similar to this one.

bubnikv, 2018-06-04:
This issue has been fixed in the 1.40.0-beta. Closing.

closed 2018-06-04 by bubnikv