prusa3d/PrusaSlicer #869: OctoPrint IP address not shown/loaded when using a custom configuration file

stelgenhof, 2018-05-08:
### Version
v1.40.0 
git rev-parse HEAD: 8eb9ddc2eb6c3f98cf28125ddbee4d233fa58616

### Operating system type + version
Arch Linux 4.16.7-1-ARCH

### Behavior
When using a custom configuration file (config.ini) that contains the settings to an OctoPrint server, the IP address is not loaded/visible in the Printer window. Even if the custom configuration file contains the correct IP address setting.

Other related settings, such as the API key, are loaded correctly.

[image]
(Note: just blurred my key :) )

stahlfabrik, 2018-05-15:
I have the same issue with the first alpha release of 1.40.

I entered octopi full DNS name "octopi.fritz.box:5000". When reloading slic3r it is gone.

bubnikv, 2018-05-17:
It is a regression issue, we  used a piece from the upstream slic3r, where the octoprint_host has been renamed to print_host. I will comment this piece of code out.

bubnikv, 2018-05-17:
This issue should be fixed with https://github.com/prusa3d/Slic3r/commit/35bf4aee7df3baee5f64b769d58d20199bb5268a

@FidelCapo Please retest with the master. Thanks.

stelgenhof, 2018-05-17:
Just compiled and tested commit 35bf4aee7df3baee5f64b769d58d20199bb5268a, but the issue still remains.

I tested with both an external configuration .ini file as well as my custom printer preset (stored in the .Slic3rPE folder).

stelgenhof, 2018-05-18:
@bubnikv Thanks for the fix! I can confirm it works now.

bubnikv, 2018-05-18:
great, thanks for testing.