prusa3d/PrusaSlicer #870: Feature Request: Insert M117 progress messages for USB/OctoPi prints

marcaurelkiefer, 2018-05-08:
When printing from SD card, progress is displayed on the LCD, this is not the case when printing via USB/OctoPi.

Since Slic3r can pretty accurately compute the print duration, wouldn't it be easy to insert progress messages via M117 command to display progress in percent (real time, not just line numbers or file size) or remaining time or whatnot.

As far as I looked into scripting, this cannot be done with a script, right?
I would love this feature!

tracernz, 2018-05-08:
Octoprint already has a couple of plugins for this:
https://plugins.octoprint.org/plugins/displayprogress/
https://plugins.octoprint.org/plugins/detailedprogress/

ManuelMcLure, 2018-05-08:
Also, if you're using Marlin and have `LCD_SET_PROGRESS_MANUALLY` set in your configuration you can use:
https://plugins.octoprint.org/plugins/m73progress/
to get the progress bar to update.

marcaurelkiefer, 2018-05-09:
But these calculations are all based on the percentage of gcode lines processed vs. lines left. And that is far from accurate. Slic3r has a far better estimation of how long a gcode command will take.

marcaurelkiefer, 2018-05-22:
@ManuelMcLure 

> Also, if you're using Marlin and have LCD_SET_PROGRESS_MANUALLY set in your configuration you can use: https://plugins.octoprint.org/plugins/m73progress/ to get the progress bar to update.

Have you tried this? Prusa Marlin does not seem to support M73...

ManuelMcLure, 2018-05-22:
> Have you tried this? Prusa Marlin does not seem to support M73...

It's definitely supported in official Marlin (I use it on my Anet A8), I can't speak for Prusa's fork.

bubnikv, 2018-08-02:
Prusa Marlin for the MK3 / MK2.5 supports M73, see the change log of https://github.com/prusa3d/Slic3r/releases/tag/version_1.41.0-alpha2

Slic3r 1.41.0-alpha2 and newer produce time estimates emitted with the new M73 code.

marcaurelkiefer, 2018-09-03:
I'm using a MK3. Love this feature, thanks for implementing it!

bubnikv, 2018-09-03:
So the original request has been satisified. Closing.

closed 2018-09-03 by bubnikv

lf-, 2018-12-03:
It has not been satisfied for MK2s, which still would benefit from this feature substantially. Needs reopening.

Also, if OctoPrint is being used with an SD card, its time estimator is garbage compared to the Slic3r one, so we really need this in Slic3r.

bubnikv, 2018-12-03:
IMHO it would be better to add support for the M73 code the same way we do
it for the MK3 or MK2.5.

lf-, 2018-12-04:
That's likely true, but I wasn't sure if the firmware is still compatible with the MK2S and nobody has bothered compiling it or if releasing an update would require substantial hacking. Would be great to see a new firmware with that feature integrated though!