prusa3d/PrusaSlicer #872: Segmentation Fault when switching View option in Preview Tab

stelgenhof, 2018-05-11:
### Version
1.40.0-alpha
git rev-parse HEAD: 0fc422544bc366dac548abc6d291c99ae76d8819

### Operating system type + version
Arch Linux 4.16.7-1-ARCH

### Behavior
Slic3r crashes when you switch the View Option of the Preview Tab to anything else but "Feature Type".

Steps to reproduce:
1. Load an STL model (any will do)
2. Slice and export the model to Gcode
3. Open the Preview Tab in the main Plater Window
4. Change the "View" option to "Speed" for example.
5. Slic3r will crash.

[image]

lukasmatena, 2018-05-11:
Thanks for reporting, it should be fixed by https://github.com/prusa3d/Slic3r/commit/d89f8128cde85f45beac212847ee0f578eab26ed.

I'll leave the issue open for now and close it later, unless there are objections from you, @enricoturri1966, @bubnikv or anyone else.

bubnikv, 2018-05-11:
Thanks, closing.

closed 2018-05-11 by bubnikv

stelgenhof, 2018-05-11:
Recompiled. Confirm the issue has been resolved. Thanks for the quick fix!