prusa3d/PrusaSlicer #873: Build fails with "Can't locate ExtUtils/Typemaps/Basic.pm in @INC..."

kortschak, 2018-05-11:
### Version
version_1.40.0-alpha a5e48cdf4d824fdec8a97d1f1569e58595d0990e

### Operating system type + version
Linux ubuntu 16.04.4

### Behavior
Attempt to build Slic3r according to the [wiki entry](https://github.com/slic3r/Slic3r/wiki/Running-Slic3r-from-git-on-GNU-Linux).

Expect (laughingly given the involvement of perl) a clean build.
Observe the following failure:
```
<snip>
[ 93%] Built target Shiny
[ 94%] Generating typemap
Can't locate ExtUtils/Typemaps/Basic.pm in @INC (you may need to install the ExtUtils::Typemaps::Basic module) (@INC contains: /etc/perl /usr/local/lib/x86_64-linux-gnu/perl/5.22.1 /usr/local/share/perl/5.22.1 /usr/lib/x86_64-linux-gnu/perl5/5.22 /usr/share/perl5 /usr/lib/x86_64-linux-gnu/perl/5.22 /usr/share/perl/5.22 /usr/local/lib/site_perl /usr/lib/x86_64-linux-gnu/perl-base .).
BEGIN failed--compilation aborted.
xs/CMakeFiles/XS.dir/build.make:101: recipe for target 'xs/typemap' failed
make[2]: *** [xs/typemap] Error 2
CMakeFiles/Makefile2:447: recipe for target 'xs/CMakeFiles/XS.dir/all' failed
make[1]: *** [xs/CMakeFiles/XS.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
```

The output of `perl Build.PL` clearly shows this package is installed.
```
<snip>
Successfully installed ExtUtils-ParseXS-3.35 (upgraded from 3.28)
1 distribution installed
ExtUtils::Typemaps is up to date. (3.35)
--> Working on ExtUtils::XSpp
Fetching http://www.cpan.org/authors/id/S/SM/SMUELLER/ExtUtils-XSpp-0.18.tar.gz ... OK
Configuring ExtUtils-XSpp-0.18 ... OK
<snip>
```

Complete build session [here](https://gist.github.com/kortschak/20aefa99ea933cd50376eb7a4318578a).

bubnikv, 2018-05-11:
The simplest fix is likely to install the missing package from ubuntu. This will likely help:

https://packages.ubuntu.com/search?keywords=libextutils-makemaker-cpanfile-perl

kortschak, 2018-05-11:
That does not solve the problem - nor did I expect it to (I'm so glad I don't use perl anymore).

kortschak, 2018-05-12:
It looks to me like the problem is that you are not setting PERL5LIB correctly prior to the build; you are installing the dependencies in .../Slic3r/local-lib/lib/perl5/ but `@INC` only includes the standard site-wide locations.

kortschak, 2018-05-12:
Confirmed by invoking `make` like so `PERL5LIB=/home/xxxx/Development/3d/Slic3r/local-lib/lib/perl5 make` with success.

kortschak, 2018-05-14:
Also, `ExtUtils::Typemaps::Basic` needs to be added to Build.PL.

bubnikv, 2018-05-14:
Would you please provide a pull request?

kortschak, 2018-05-14:
Done. This fixes the dependency issue, but I don't use cmake, so I'm not sure what is not working there - from reading the docs, what you have written looks like it should work.

kortschak, 2018-05-15:
I've looked though this and the cmake PATH and PERL5LIB env vars are not plumbed through to where they are used. From what I understand, the way they are being expected to work here is not how they actually work. It looks like the expectation is that seting `ENV{VAR}` results in an export-like situation. This is not the case. I've spent some time fixing this plumbing, but it's not clear to me how it should be done, and the combination of cmake and perl's approach to error reporting make this very difficult (Fortran and even C++ error reporting are better).

The central issue is in how CMakefile.txt sets up the include - AFAICS it should just make a path and hand that to perl with a `-I` in xs/CMakeLists.txt (this part is trivial). The handling of `$PATH` is harder since there are a couple of multiples of calls to xsubpp and xspp that need to know a path.

On a meta note, this appears to have persisted for as long as it has because the system-wide presence of various modules and executable. There should probably be some kind of testing on clean hardware set up to catch these things - there are many CI services that would help here.

vojtechkral, 2018-05-18:
@kortschak 

> From what I understand, the way they are being expected to work here is not how they actually work. It looks like the expectation is that seting `ENV{VAR}` results in an export-like situation. This is not the case.

Indeed, or, to be more specific, the variable is exported when using `execute_process`, but not when using `add_custom_target` and `add_custom_command` for some reason. I'd say this is at least a documentation bug in CMake, as the documentation doesn't mention this at all.

However, according to [StackOverflow](https://stackoverflow.com/questions/35029277/how-to-modify-environment-variables-passed-to-custom-cmake-target/35032051), the way to export an env var to a custom target/command is:

    add_custom_target(newtarget ${CMAKE_COMMAND} -E env NAME=VALUE command ...)

Basically CMake provides a portable `env` utility. We could use that.

I'm having a look at your pull request. I'm not very knowledgable about Perl. Which would you say is preferable, using the `PERL5LIB` env var or passing the `-I` option? Because apparently being able to export the env var using the above code we can do either.

Thank you!

> On a meta note, this appears to have persisted for as long as it has because the system-wide presence of various modules and executable. There should probably be some kind of testing on clean hardware set up to catch these things - there are many CI services that would help here.

Oh, absolutely. It's just a matter of finding the cycles to do that.

kortschak, 2018-05-18:
>I'd say this is at least a documentation bug in CMake, as the documentation doesn't mention this at all (Edit: I created an issue over at CMake bugtracker).

Agreed. Thanks for doing that.

>I'm having a look at your pull request. I'm not very knowledgable about Perl. Which would you say is preferable, using the `PERL5LIB` env var or passing the `-I` option? Because apparently being able to export the env var using the above code we can do either.

I've been fortunate enough not to spend much time with perl over the past decade. But in this situation, I think passing an include into perl with `-I` is more appropriate. There is a single location path to add, rather than changing an entire environment. Others my disagree though.

If you want, I can change to the approach you found at StackOverflow.

vojtechkral, 2018-05-18:
That's ok, the `-I` approach is fine with me as far as I can tell.

The pull request LGTM, I'd merge it, but I'll wait for @bubnikv 's confirmation he's ok with it :-)

closed 2018-09-03 by vojtechkral