prusa3d/PrusaSlicer #874: Weird zoom change after changing the View orientation

Panayiotis-git, 2018-05-11:
### Version
1.39.2+win64

### Operating system type + version
Windows 7 - 64 bit

### Behavior
When using the View shortcut keys, the view orientation changes without changing the current zoom. If I rotate or move the bed it still keeps the current zoom.

But, if then I scroll the mouse wheel to change the zoom or I move a part on the bed or add a new part, the zoom instantly changes to a default value.
The second case, after moving a part, is worst because if I have many parts and I have zoomed in to align them, then the automatic zoom out creates frustration.

### Expected Results
The automatic zoom should not happen.

enricoturri1966, 2018-05-14:
Zoom logic has been partly modified with: https://github.com/prusa3d/Slic3r/commit/f8b1dc550685a8ca76c4d7a810151eabf08db2c1
Now when changing the view, the zoom is set to the volumes' extent.
Also, two new hotkeys have been added:
v -> zoom to volumes' extent
b -> zoom to bed's extent

enricoturri1966, 2018-05-15:
Automatic zoom to volumes' extent when changing view removed with: https://github.com/prusa3d/Slic3r/commit/328eba56102329c415e130a24977b54277412cd2

bubnikv, 2018-05-17:
Fixed, it will be part of the beta.

closed 2018-05-17 by bubnikv