prusa3d/PrusaSlicer #875: Too many loops on skirt

Carib3D, 2018-05-11:
### Version
Prusa edition 1.39.1

### Operating system type + version
Windows 10, 64 bit

### Behavior
Applies too many loops on the skirt.
Set to 1 loop at 5mm, minimum 150mm (tried a few variations, but didn't seem to help). Single perimeter is over 200 mm long so should only print 1 loop

_Is this a new feature request?_
No

3Delights, 2018-05-18:
Confirmed the problem is also in 1.40.0-alpha1+win64.

lukasmatena, 2018-05-29:
Hi.
As far as I can tell, the behaviour is correct. The tooltip for 'Minimal extrusion length' explains that it is actually required length of the FILAMENT, not of the extrusion line on the bed. This also makes more sense, because the purpose of having this setting is to ensure proper purging of the nozzle. If it meant length of the line, one would e.g. get less actual purging for smaller layer heights. Does it make any sense?

You are right, however, that the name of the setting is very unclear and counterintuitive. We will consider changing it to 'Minimal filament extrusion length' or something similar, that might indeed be better (@YuSanka). Any other suggestions and ideas are welcome.

I'll leave this issue open until we actually commit the change. Thanks for the feedback.

Carib3D, 2018-05-30:
Okay, understood. In my case, I am not using these loops to purge the nozzle, which I have never felt is a major issue but rather putting down a number of loops that allows me to gauge if the Z height is correct and make any adjustments. So I'm looking at the skirt from a different perspective.

bubnikv, 2018-06-04:
@YuSanka is going to change the wording in the 1.40.0-final to explain that is the filament extrusion length, not the extrusion length. Closing.

closed 2018-06-04 by bubnikv

stemaDev, 2020-12-02:
I had to search for this, it was still not clear, maybe adding something like "raw filament (not printed) length" should be in it's description