prusa3d/PrusaSlicer #877: Should wait for bed temperature in sequential printing mode

wavexx, 2018-05-12:
### Version
1.39.2

### Operating system type + version
Debian unstable

### Behavior
Enable "Complete individual objects" in Print settings -> output options -> sequential printing.
In Filament settings -> Filament -> bed select a different temperature for first layer and other layers (I use 65 and 55 normally).

Load 2 objects onto the plate -> print.

After completing the first object, the temperature of the bed is set again to first layer as it should, but the print starts immediately without waiting for the temperature to be reached.

This is suboptimal. I use sequential printing when lots of small objects need to be printed fast, as there are no long travel moves between the objects. However, for small objects the first layer temperature is also critical for the piece to stick.

Ideally we would need to stop the travel move towards the second object halfway and wait until the bed temperature is reached before continuing to print.

bubnikv, 2018-05-17:
Slic3r does not wait for the temp changes on purpose. I have rewritten this piece of code, but I just kept the logic equal to the original slic3r. I thought there is some reason behind it.

Should Slic3r wait just for the bed to reach its temperature, or should it wait for both the bed and extruder?

wavexx, 2018-05-17:
There are several reasons I could think of, but the big one is oozing. If you wait to raise the bed temp by 10c, it might take several seconds in which the nozzle will probably start oozing somewhere.

If there was a per object skirt (I can think of #678), this wouldn't be an issue, as it would wipe the nozzle.

bubnikv, 2018-05-17:
Now what do you want? :-)

Are you aware, that if you place the temp control G-codes into the "between objects" custom G-code, Slic3r will not generate its own? So you may easily work around this issue, and you may add some priming as well.

wavexx, 2018-05-17:
You're right ;). I thought about oozing only after you mentioned of possible issues.

What I do currently is place each object independently on the plate, slice it, and concatenate the produced gcode with some preprocessing (mainly to avoid startup gcode homing the nozzle). This gives me also individual skirts, but the whole process is not as automatic as I would like.

Ideally, this is what I'd consider to be a true "complete individual objects" mode.

Effectively, if you don't wipe the nozzle, waiting for bed temperature might be a bad idea.

rennne, 2019-01-23:
What about rising the bed temperature maybe 20 seconds before the object form before is finished.
Then when it travels to the next print, the necessary temperature is already reached.

I have similar issues -> small prints falling of the bed because the first layer was printed onto a too cold bed, when doing sequential printing 

I print tiny mechanical parts, thereby good results are reached with 65° First Layer  to make the parts stick better and 30° for the rest, to get exact overhangs even near the bed.

wavexx, 2019-01-23:
I didn't think of this: it's a perfect idea. Issue an M140 in anticipation. You can also estimate the time is takes for the bed to heat up, so that the M140 is sent in the reasonable time frame.

rennne, 2019-01-26:
I think as a manual workaround it could work, to move the line with M140, 500 lines up in the gcode-file.

wavexx, 2019-01-27:
On Sat, Jan 26 2019, rennne wrote:
> I think as a manual workaround it could work, to move the line with
> M140, 500 lines up in the gcode-file.

Definitely doable also using a post-processing script. However to do it
properly you need to know in advance how long the bed takes to reach the
target temperature from the current and roughly the time required by
each layer in order to place the statement at the correct spot.

I have three different printers at my disposal and I'm pretty sure you
can approximate almost perfectly the heating time with just a
log(scalar). The ambient temperature makes almost no change as I
confirmed this experimentally (I often print in an enclosure). This
value can easily pre-set for a printer in the default config.

To move the heating to the right spot though you need to work backwards
through the layers accumulating the printing time.

For proper control though you'd also need a way to set the printing
order of each element. On the MK3 this is especially important, as the
extruder is quite big. But if you print each object in increasing Y
order, then you can still pack quite a considerable number of objects on
the build plate.

rennne, 2019-01-27:
I played a lot around with manually editing the gcode. Manually it is quite difficult to find the right position in the gcode. So a lot of playing around. 
A good starting point is  to look at the number of lines and divide them by the number of minutes.

1 minute before the change between the objects is a good value. 
It also seems that the heat takes a bit of time to travel through the detachable build plate to the upper surface. So for that there should be added some extra time.

foreachthing, 2020-05-12:
PS 2.3.0 alpha (maybe prior versions as well) has "between objcts G-Code" in `Printer Settings`.
Just add `M190 S[first_layer_bed_temperature]`. [M190 - Wait for Bed Temperature](https://marlinfw.org/docs/gcode/M190.html).

MechanicalCat, 2022-04-24:
When sequentially printing objects in PET-G, the second object does not properly attach to the bed because the bed-temp and extruder-temp remain too high after the previous object. (Temperatures for the FIRST layer when printing PET-G are COLDER than all the other layers - this is needed to get a good bond to the bed). I repeatedly get print failures (second object detaching from the bed during printing) because of this.
M109 and M190 (inserted manually into the g-code) do work better. But even better would be M109, M190, G4 S60... because otherwise the machine STILL restarts before the temps are an exact match (bed temp about 3C too high).
If oozing is an issue, then maybe another intro-line (adjacent to the second object, so as not to foul the first object) would help?

MechanicalCat, 2022-04-24:
A better solution for this whole problem would be to print ALL the first-layers of ALL the objects at the start of the print-run. Then print all the remaining layers (sequentially). Maybe that could be an option added to Sli3er?

bluerider098, 2024-09-09:
I've had similar issues, but I dont think it's temperature related.  I'm my case the first layer temperature is only 5°C different between the first layer and the subsequent layers.  

I think the issue is related to the bed mesh and how the shape of the bed changes as the frame and bed absorb heat.  Having other printed objects on the bed will affect the shape as well. 

I've had succesd with preheating for 60 minutes to get a good heat soak before the initial bed mesh and then lowering the live Z by .01.  This leaves the first layer a bit squished on the first part, but it looks good on the rest.

I've spoken with others that have had similar issues and they said the nozzle was way too far off the bed for the subsequent parts as well. 

So either the bed shape is changing to much between objects, or PS is not properly applying the bed mesh to subsequent parts. 

It might also not be a bad idea to keep your first layer temperature the same as other layers.  With a proper nozzle offset (automatic depending on your printer) for a good first layer and the correct print sheet, I can't see a few degrees C making or breaking first layer adhesion.

closed 2026-09-29 by prusaslicer-bot[bot]