prusa3d/PrusaSlicer #879: Raft with layerheights below 0.2mm results in mid-air-print

wildblue, 2018-05-13:
### Version
_Version of Slic3r used goes here_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Windows

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
Printing with a raft does not Work correctly, especially with layerheights below 0.2mm. The height of the First Thin layer does result in printing mid Air.

* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_
No

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_
Any

FidelCapo, 2018-05-14:
Thank you for the report. Unfortunately, I am not able to reproduce the problem you have described. Can you please provide configuration of your settings? (File - Export config ..) Thank you

bubnikv, 2018-05-17:
I suppose the problem is, that there is no room for the supports to guarantee the gap between the supports.

bubnikv, 2019-01-29:
No response, closing.

closed 2019-01-29 by bubnikv