prusa3d/PrusaSlicer #881: Crashes on Startup

Thoin, 2018-05-13:
### Version
* 1.39.2 x64

### Operating system type + version
* Windows 10 Pro
* Version 1709
* OS Build 16299.431

### Behavior
* _Describe the problem_
When opening Slic3r PE, the UI opens, then immediately it grays out and Windows shows the dialog box that says Slic3r has stopped working.

* _Steps needed to reproduce the problem_
The only new piece of software that might have interfered with Slic3r would have been Meshmixer. I only noticed that Slic3r wasn't working because I installed Meshmixer and it crashed on startup and I was going to use Slic3r to make the same changes to my files. I have since uninstalled Meshmixer. I also uninstalled and reinstalled Slic3r PE several times in different ways. I had Slic3r installed and working for several months then stopped working unexpectedly.

  * _Screenshots from __*Slic3r*__ preview are preferred_
[image]

### Notes
* I am new to Slic3r so I do not know what kind of logging or troubleshooting tools are available in the software. Please let me know what else I can do to get you the necessary information you need to troubleshoot this issue. Thank you.

bubnikv, 2018-05-16:
I guess there is an issue with your graphics card drivers. Do you still have the issue? What kind of graphics card? Will an update of your graphics card help?

jpatenet, 2018-05-16:
I got also the same problem several times without use anithing else at the same time. The last time, Slic3r stopped when I ask to record the gcode file.