prusa3d/PrusaSlicer #882: Overhang perimeter at high speed instead of setting, only on a single side (steep overhang)

LoganFraser, 2018-05-13:
### Version 
1.39.2+win64

### Operating system type + version
Windows 10 Pro 64bit (up to date)

### Behavior
I've selected 40 as perimeter speed, with 50% for external perimeters. On one side of this object (with the steepest overhang), the speed for the external perimeter is double instead of half, resulting in serious waving of prints. 

The speed is correct on all other perimeters. I believe the last line isn't being detected as a perimeter.

#### STL/Config (.ZIP) where problem occurs

[CarcasaRRD_tapa.zip](https://github.com/prusa3d/Slic3r/files/1998964/CarcasaRRD_tapa.zip)

LoganFraser, 2018-05-14:
This only appeared to happen up to around slice 7 or 8 and then expected speed for the overhang perimeter resumed.

bubnikv, 2018-05-14:
@FidelCapo reviewed your report. According to him, the perimeters printed faster are classified as overhang perimeters, therefore the bridging speed and bridging flow apply. Does this explain what you are seeing?

LoganFraser, 2018-05-14:
That seems counter-intuitive. A bridge is from a solid point to an adjacent point with no attachment between the points.  An overhang is attaching continuously along one side.  The former wants to be fast to keep drooping to a minimum, but the latter wants the opposite. Also, the ramp is the same angle on the first 20 or so slices and it only traveled at a higher speed for maybe 6 slices and then continued at the slower rate. I have bridging speed set to 60mm/sec but it appeared to be travelling faster than that.

bubnikv, 2018-05-17:
Do you mean the green extrusions?

[image]

The screenshot has been taken from 1.40.0-alpha1
https://github.com/prusa3d/Slic3r/releases/tag/version_1.40.0-alpha1

LoganFraser, 2018-05-17:
Yes. So it probably is the bridge code being used. :( That means no real easy to fix it. I've seen numerous threads requesting a seperate bridge and overhang settings. I'll just have to add my request there as well.

rtyr, 2020-03-24:
I believe we can close this. Request for separate print speed setting for overhangs is discussed here https://github.com/prusa3d/PrusaSlicer/issues/242.

closed 2020-03-24 by rtyr