prusa3d/PrusaSlicer #886: Slic3r 1.40.0-aplha1 often does not show various visualizations

stahlfabrik, 2018-05-14:
### Version
Slic3r 1.40.0-aplha1

### Operating system type + version
MacOS High Sierra

### Behavior
I do not know how to reproduce. The first time I looked at the new visualizations, they would not show up. Just the ordinary yellow sliced layers like in the last stable release version. Then I sliced another file and the visualizations (of speed, flow rate etc.) showed in their full glory. I then sliced a third file and the visualizations did not show - again. So how I could I make sure to always get the visualizations? I attach a pic of the last try.

#### STL/Config (.ZIP) where problem occurs
Model is Mooncity 2.0  - stock/unaltered
[image]

Petclaud, 2018-05-14:
Yep having the same issue. If I send the file to the printer (to Octoprint) the visualisations are there (the same issue and "resolution" I have with the printing time - this also only shows up after sending to the printer).

stahlfabrik, 2018-05-14:
Aha! Do you say after I send it to OctoPrint the visuals come to live? I will try that tomorrow! 

That would be a workaround until this is fixed

Panayiotis-git, 2018-05-14:
You have to first export the G-code. Then the visualizations are available.

vernieg, 2018-05-14:
Showing different colors visualization after exporting G-code the first time is ok. Deleting the first sliced file and loading again to slice another file, visualization no longer work completely after exporting the G-code, internal perimeter is gone. Deleting and loading another file to slice then export, it works fine as well for the rest of different files after that. This happens on  the linux version. Somehow after the first slicing, the problem only shows once on the second exporting of G-code then no more. I was using only two different stl files during testing to replicate the problem.

stelgenhof, 2018-05-14:
@stahlfabrik @Petclaud The Preview tab is a new feature that can visually display the generated gcode in various ways (by speed, height, etc.) As such, this tab will only show something when gcode is actually generated, either by choosing "Export gcode" or "Send to Printer".

Only slicing your 3D Model will not render any results in the Preview tab. You need to generate the gcode. The same goes for the Slicing information (like Estimated print time). That is also based on the generated gcode.

So, it's not an issue/bug. It's by design :)

vernieg, 2018-05-15:
[image]
Now I understand, somehow the Perimeter was toggled off that is why it was not shown. But I did not toggled it or explore in that area when testing earlier. Anyway, this new visualization is really nice to have and thank you so much for it.

bubnikv, 2018-05-15:
> So, it's not an issue/bug. It's by design :)

We will certainly make that automatic in 1.41, but it will require large
changes in the architecture, so we did not want to delay the release.

> Now I understand, somehow the Perimeter was toggled off that is why it
was not shown. But I did not toggled it or explore in that area when
testing earlier.

We have some issues with the wxWidgets on some architectures. It may be,
that wxWidgets toggled the perimeters checkbox while it should not have to.
We will certainly fix that one day, ideally by contributing the change into
the official wxWidgets, but it will be much easier once the complete Slic3r
gets rid of Perl.

Petclaud, 2018-05-15:
`We will certainly make that automatic in 1.41`

That would be great. thx.

stahlfabrik, 2018-05-15:
Great that you will make it automatic - I hope time estimation as well by the way:-) from a user's perspective it is not obvious why it should not be displayed automatically after the "slice" is done

bubnikv, 2018-05-17:
> Should I close the issue because it is already on the 1.41 todo list?

It is on the 1.41 list.

Closing.

closed 2018-05-17 by bubnikv