prusa3d/PrusaSlicer #889: Retraction

Itox001, 2018-05-15:
### Version
Slic3r PE 1.40 Alpha 1

### Operating system type + version
Windows 10 64 bit Home Edition

### Behavior
Good job on the Gcode visualizer, it is awesome and it can certainly help debug certain prints, watching how your print artifacts match up with the visualization of speed/flowrate/retractions/deretractions is an awesome way to know what is going on and what settings to tweak.

I found a bug in the visualization, when retraction/deretraction/travel moves checkboxes are activated and you are watching them in the model, if you move the layers slider to visualize different layers, suddenly retraction/deretraction/travel moves dissapear, and won't appear again until after you return the layers slider above certain threshold.

Here in this model the threshold is layer 50:

[image]
On layer 87 everything is fine

[image]
On layer 50 it still visualizes correctly

[image]
Below layer 49 none of the options are visualized

This benchy has a 0.25 layer height, compared to 0.2 on the one above. In this one the threshold is layer 40, just below the 10 mm mark, since layer 50 on the 0.2 LH benchy is also the 10 mm mark, I conclude that for some reason this features stop rendering if layers above 10 mm are hidden:
[image]
 

BTW, the shells checkbox seems to do nothing. Is that intended?

bubnikv, 2018-05-17:
@enricoturri1966 is the author of the new G-code preview. Would you please look into this issue? Thanks.

By the way, I am seeing the following output on the console, which may be an indication of the broken "Shells" visualization?

```
Use of uninitialized value $z_low in sprintf at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 439.            
Use of uninitialized value $z_low in sprintf at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 439.            
Use of uninitialized value $z_high in sprintf at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 440.           
Use of uninitialized value $z_high in sprintf at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 440.           
Use of uninitialized value $z_low in subtraction (-) at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 445.    
Use of uninitialized value $z_high in addition (+) at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 445.      
Use of uninitialized value $z_low in sprintf at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 439.            
Use of uninitialized value $z_low in sprintf at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 439.            
Use of uninitialized value $z_high in sprintf at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 440.           
Use of uninitialized value $z_high in sprintf at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 440.           
Use of uninitialized value $z_low in subtraction (-) at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 445.    
Use of uninitialized value $z_high in addition (+) at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 445.      
Use of uninitialized value $z_low in sprintf at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 439.            
Use of uninitialized value $z_low in sprintf at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 439.            
Use of uninitialized value $z_high in sprintf at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 440.           
Use of uninitialized value $z_high in sprintf at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 440.           
Use of uninitialized value $z_low in subtraction (-) at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 445.    
Use of uninitialized value $z_high in addition (+) at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 445.      
Use of uninitialized value $z_low in sprintf at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 439.            
Use of uninitialized value $z_low in sprintf at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 439.            
Use of uninitialized value $z_high in sprintf at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 440.           
Use of uninitialized value $z_high in sprintf at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 440.           
Use of uninitialized value $z_low in subtraction (-) at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 445.    
Use of uninitialized value $z_high in addition (+) at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 445.      
Use of uninitialized value $z_low in sprintf at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 439.            
Use of uninitialized value $z_low in sprintf at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 439.            
Use of uninitialized value $z_high in sprintf at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 440.           
Use of uninitialized value $z_high in sprintf at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 440.           
Use of uninitialized value $z_low in subtraction (-) at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 445.    
Use of uninitialized value $z_high in addition (+) at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 445.      
Use of uninitialized value $z_low in sprintf at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 439.            
Use of uninitialized value $z_low in sprintf at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 439.            
Use of uninitialized value $z_high in sprintf at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 440.           
Use of uninitialized value $z_high in sprintf at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 440.           
Use of uninitialized value $z_low in subtraction (-) at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 445.    
Use of uninitialized value $z_high in addition (+) at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 445.      
Use of uninitialized value $z_low in sprintf at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 439.            
Use of uninitialized value $z_low in sprintf at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 439.            
Use of uninitialized value $z_high in sprintf at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 440.           
Use of uninitialized value $z_high in sprintf at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 440.           
Use of uninitialized value $z_low in subtraction (-) at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 445.    
Use of uninitialized value $z_high in addition (+) at D:/src-perl/Slic3r-master/lib/Slic3r/GUI/Plater/3DPreview.pm line 445.      
```

enricoturri1966, 2018-05-17:
Shell visualization fixed with: https://github.com/prusa3d/Slic3r/commit/e13564cb190458a818f6ad40dc8705c9c6ae8a5e

Itox001, 2018-05-20:
Here is both the 3MF file and the exported config.

Hope you can figure something out with this one.

[Evidence.zip](https://github.com/prusa3d/Slic3r/files/2020023/Evidence.zip)

bubnikv, 2018-06-04:
@FidelCapo Would you please test this issue and close it if it is working?

FidelCapo, 2018-06-04:
Tested and working ok in 1.40.0-beta. I'm closing the issue.

closed 2018-06-04 by FidelCapo