prusa3d/PrusaSlicer #890: 1.40.1 alpha1 - OS X real G-code preview Feature check boxes not always working

TheZeroBeast, 2018-05-15:
1.40.1-A1

### Operation System
OS X Mac

### Behavior
When navigating visability of features enabled in the real G-code preview on OS X the check boxes don't respond to click action as would be expected. If the text beside the check box is selected the check box updates and changes although working this could be cleaned up a little for a more streamlined user experience.

Thank you for the good work so far, keep it up.

Rob

wabbitguy, 2018-05-16:
Well that procedure would have never entered my mind so, thank you! 

Exported the GCODE and it all showed up and works pretty slick. Could me save from having to use code.ws in future.

wabbitguy, 2018-05-17:
@lukasmatena for whatever reason my brain skipped through first paragraph that specifically said "export" and devoured the second paragraph with rapt interest instead....no doubt the programmer in me guilty "optimizing" documentation...:-)