prusa3d/PrusaSlicer #891: Cannot set skirt-to-brim distance to 0

fpikus, 2018-05-16:
### Version
1.39.2

### Operating system type + version
Windows 10

### Behavior

I'm setting skirt distance to 0. According to the popup, the skirt should be attached to the brim. But this does not happen, it stays few millimeters away from the brim. The skirt distance is honored up to some small value about 2mm: If I set it to 20mm, it's 20mm away from the brim. 10mm, no problem, 10mm away. At about 2mm it stops moving closer to the brim. The value is accepted, down to 0, but the gap between the skirt and the brim remains. I would like the skirt to be the outer loop of the brim, with zero gap (so the skirt adhered to the bed better)

vernieg, 2018-05-16:
Skirt not sticking better to the bed can be solved by moving the bed closer to the nozzle, this is assuming the bed is clean and not oily.  Skirt as part of the brim is like having additional line to the brim while skirt is turned off.

fpikus, 2018-05-16:
There is a big difference between an additional loop of the brim and the skirt: skirt can be more than one layer tall. In particular, the skirt can be as tall as the part itself. This reduces drafts and keeps the print at a constant temperature. The problem is, the tall skirt does not adhere to the bed very well. You would not print a very thin wall directly without a brim, same here. Tall skirt needs to be attached directly to the brim. 
The pop-up help bubble on the skirt distance says that it should work, it explicitly says to set the distance to 0 to connect the skirt to the brim. But it doesn't work.

lukasmatena, 2018-05-22:
Thanks for your feedback, we looked into it. I'm closing this issue as a duplicate of https://github.com/prusa3d/Slic3r/issues/724, where you can read about the recent update. Post any other comments there, please.

closed 2018-05-22 by lukasmatena