prusa3d/PrusaSlicer #892: Export mega file or pkg

mvasilakis, 2018-05-16:
### Version
1.40.0 alpha 1

### Operating system type + version
Mac OS 10.13.4

### Behavior
* When troubleshooting issues (such as the [Inconsistent Extrusion thread](https://github.com/prusa3d/Prusa-Firmware/issues/602#issuecomment-389022163) ) We need to export the G-Code, STL and slicer settings for other users and Prusa support. It would be a huge time saver if we could export a slic3r file or pkg for support so that slic3r can import the STL, Print settings, Filament, Printer settings and last generated g-code to view, manipulate and test print.
* _Expected Results_
Export a stand alone file or pkg that slic3r can import that contains all the settings (With the exception of Octoprint settings), STL and last g-code generated into one file.
* _Actual Results_
Currently all files need to be packaged separately for trouble shooting. Most users only export the g-code and that takes allot of time to browse to see what may be wrong with the users settings.

_Is this a new feature request?_
YES

bubnikv, 2018-05-16:
funny, the 1.40.0-alpha1 relase log contains a section on such an archive.
3mf and amf formats could now encompass all the configs. The g code is
missing though, but in most cases it is not needed as it could be generated
from the other data.

mvasilakis, 2018-05-16:
🙄 And I read the log. Twice. Thanks, I'll re-read and mess with the archive formats in the alpha.

closed 2018-05-16 by mvasilakis