prusa3d/PrusaSlicer #893: 1.40.0 Alpha 1 - Right tools drawing too narrow

albegouws, 2018-05-16:
Version
1.40.0-alpha1+win64

### Operating system type + version
Windows 10 Pro 1709

### Behavior
Screen drawing - right hand side toolbar too narrow.  This doesn't leave enough space to view all preset names and fields.
Screen resolution 1920x1080
See screenshots

[image]
[image]

Panayiotis-git, 2018-05-16:
In your previous installation, you have selected the background processing. At the new installation you have not. So, the button 'Slice now...' is visible and this stretches the right column, so the drop down boxes are drawn OK.

The problem still exists. Just select again the background processing.

A simple solution could be to have the right column width adjustable with a splitter.

YuSanka, 2018-05-17:
Good catch :) Thanks! Info boxes were seted minimum right column width before. But after placed thay to the Scrolled Window I forgeted to set minimum width to the main sizer. Fixed bug is in the master now.

bubnikv, 2018-05-17:
Looks good to me. It will be part of the beta. Thanks, closing.

closed 2018-05-17 by bubnikv