prusa3d/PrusaSlicer #896: bed_size crashes slic3r

repetier, 2018-05-17:
### Version
1.40.alpha

### Operating system type + version
Windows

### Behavior

If you have old printer configurations with
bed_size = 175,175

instead of 

bed_shape = 0x0,100x0,100x100,0x100

slic3r will crash as soon as you select that profile. Not sure when that value was changed, but none the less it should be parsed correctly, ignored or be converted and not crash the slicer.

bubnikv, 2018-05-17:
Would you please attach your config or any data to test with?
Thanks.

repetier, 2018-05-17:
Sure, use this printer config to make it crash once you select it.

[Ordbot.ini.zip](https://github.com/prusa3d/Slic3r/files/2012112/Ordbot.ini.zip)

Once you replace the line bed_size with bed_shape and right values it will not crash any more.

bubnikv, 2018-05-17:
Fixed with https://github.com/prusa3d/Slic3r/commit/e3d84407e0412c2fcc9e81dbb540b2872a37300b
This is quite an old bug though.
Thanks for reporting.

closed 2018-05-17 by bubnikv