prusa3d/PrusaSlicer #898: 1.40.0 alpha1 hanged due to error

gdombiak, 2018-05-18:
### Version
_1.40.0-alpha1+linux64-full-201805141245_

### Operating system type + version
_Ubuntu 16.04_

Running from command line (downloaded compressed format). Using: ./slic3r --gui

### Behavior
I was checking out the new alpha version. I clicked on the Print settings tab and started to go over each option listed on the left menu. I wanted to compare differences between preset FAST and Chris MK3 PLA v3.ini. As I got to the 3rd or 4th menu item suddenly the UI froze. I saw this in the command line

> Creating main frame...
>
>(slic3r.pl:5485): Gtk-CRITICAL **: IA__gtk_color_button_set_color: assertion 'color != NULL' failed
>  
>(slic3r.pl:5485): Gdk-CRITICAL **: IA__gdk_window_thaw_updates: assertion 'impl_window->update_freeze_count > 0' failed
>
>(slic3r.pl:5485): Gdk-CRITICAL **: IA__gdk_window_thaw_updates: assertion 'impl_window->update_freeze_count > 0' failed

Weird thing is that I can still click on Filament and Printer Settings but I cannot go back to Print Settings tab.

_Is this a new feature request?_ NO

bubnikv, 2018-05-18:
Would you please try to reproduce the issue and try to find a quickest sequence for us to be able to reproduce it? Thanks.

gdombiak, 2018-05-18:
Happened again. Seems to be easy to reproduce. This is what I did:

1. Started Slic3r from command line using: ./slic3r --gui
1. Clicked on Print Settings then 
1. Clicked on Layers and perimeters then
1. Clicked on Infill then on Skirt then on Support then on Speed then on Multiple Extruder then
1. clicked on Advanced and kaboom!

This printed to the console:

> (slic3r.pl:9082): Gdk-CRITICAL **: IA__gdk_window_thaw_updates: assertion 'impl_window->update_freeze_count > 0' failed

Before that entry I had this in the console:

>gaston@gaston-Inspiron-N7010:~/Downloads/Slic3rPE-1.40.0-alpha1+linux64-full-201805141245$ ./slic3r --gui
>Creating main frame...
>
>(slic3r.pl:9082): Gtk-CRITICAL **: IA__gtk_color_button_set_color: assertion 'color != NULL' failed
>
>(slic3r.pl:9082): Gdk-CRITICAL **: IA__gdk_window_thaw_updates: assertion 'impl_window->update_freeze_count > 0' failed

Let me know if you need more info or version of other libraries that I installed.

bubnikv, 2018-05-18:
@YuSanka could you reproduce it? Would you please try to fix it? Thanks.

bubnikv, 2018-05-18:
@gdombiak We cannot reproduce your issue. Would you please pack and attach your Slic3rPE directory with your profiles for us?

lukasmatena, 2018-05-18:
After some research I managed to reproduce the problem. It is apparently a bug in Ubuntu related to their "Overlay" scrollbars. Similar issue was reported here http://forums.codeblocks.org/index.php?topic=20771.0 (and in several threads this one links to).

I doubt there is anything we can do to fix it. However, a remedy exists: install and run unity-tweak-tool and set scrollbars to "Legacy". It will likely be ok then. You will of course have to part with the "improved" scrollbars, but that is not a big price. They are not very good to use anyway.

Please, let us know if that solved the problem.

gdombiak, 2018-05-18:
Excellent. Will give it a try when I'm back from work and report back. 

Thanks for your time and help.
Gaston

gdombiak, 2018-05-19:
Confirmed. That was the issue. I had to follow this instructions: http://www.debugpoint.com/2016/05/how-to-install-unity-tweak-tool-in-ubuntu-16-04-lts-xenial-xerus/ to make it work on Ubuntu 16.04.

After I went to 'legacy scrolling', I now only get this error in the console. I no longer have this issue but you might be interested in taking care of this inoffensive error:

>gaston@gaston-Inspiron-N7010:~/Downloads/Slic3rPE-1.40.0-alpha1+linux64-full-201805141245$ ./slic3r --gui
>Creating main frame...
>
>(slic3r.pl:3937): Gtk-CRITICAL **: IA__gtk_color_button_set_color: assertion 'color != NULL' failed

Will close this issue. Let me know if you want me to open one to document "known issues" for future people that may run Slic3r on Ubuntu 16.04.

Thanks for your quick help!!! Much appreciated.
Gaston

closed 2018-05-19 by gdombiak

bubnikv, 2018-05-19:
I think we should maintain a wiki page with known issues and their
workarounds. It would be great if you could help.

gdombiak, 2018-05-19:
Sure thing. Happy to help. Can I use the wiki of this repo? Or point me to the wiki that you guys use. 

Thanks 
Gaston

gdombiak, 2018-05-20:
Created this [wiki page](https://github.com/prusa3d/Slic3r/wiki/Known-Issues-and-Solutions)

bubnikv, 2018-06-12:
Fixed in Slic3r 1.40.0 final release.