prusa3d/PrusaSlicer #899: Load config is lost after restart

gdombiak, 2018-05-18:
### Version
_1.40.0-alpha1+linux64-full-201805141245_

### Operating system type + version
_Ubuntu 16.04_

Running from command line using: ./slic3r --gui

### Behavior
1. Start Slic3r from command line
1. Load Config Chris MK3 PLA V3.ini
1. Close app
1. Start Slic3r again. Imported config is gone

If I go into any of the imported settings and click save (even without changing anything) then that particular setting remains after restart

_Is this a new feature request?_ NO

bubnikv, 2018-05-18:
>    2. Load Config Chris MK3 PLA V3.ini
>    3. Close app
>    4. Start Slic3r again. Imported config is gone

Actually this is how it is supposed to work. The config is loaded, not
imported. If you want to import it, you have to copy the separate print,
filament, printer ini files into their particular directories. Indeed, the
config.ini does not contain the names of the included print / filament /
printer profiles as of now, so there is not a good way to name the profiles
when storing them into print / filament / printer directories.

The "Load config bundle" works differently, it actually installs the print
/ filament / printer profiles into the user directory. I suppose there is a
room for improvement here, at least to make the naming of the actions more
consistent.

bubnikv, 2018-05-18:
Let's keep the ticket open, we will discuss internally how to improve the wording.

tobymurray, 2018-10-24:
I believe I was confuse by this in #1347. The `config.ini` is actually a "config bundle"? Is there any way to use these config bundles through the command line? If not, any suggestions on how to slice from command line with the relevant config bundle entries?

rtyr, 2021-01-20:
I believe this can be closed now.

closed 2021-01-20 by rtyr