prusa3d/PrusaSlicer #900: Application crashes if no zenity, Xdialog or Kdialog present in system

dlozinski, 2018-05-18:
### Version
Prusa Edition 1.40.0-alpha1
(Slic3rPE-1.40.0-alpha1.linux64-full-201805141245.AppImage)

### Operating system type + version
Lubuntu 16.04

### Behavior

Application crashes immediately with no gui error message if `libglu1-mesa` package is not installed.

When running in terminal it prints the following:

`zenity, kdialog, Xdialog missing. Skipping /tmp/.mount_Slic3rXe61ro/usr/bin//slic3r.wrapper.
Can't load '/tmp/.mount_Slic3rXe61ro/usr/bin/lib/site_perl/5.22.0/auto/Slic3r/XS/XS.so' for module Slic3r::XS: libGLU.so.1: can't open shared object: No such file or directory at /tmp/.mount_Slic3rXe61ro/usr/bin/lib/site_perl/5.22.0/XSLoader.pm line 70.
 at /tmp/.mount_Slic3rXe61ro/usr/bin/lib/site_perl/5.22.0/Slic3r/XS.pm line 22.
Compilation failed in require at /tmp/.mount_Slic3rXe61ro/usr/bin/lib/site_perl/5.22.0/Slic3r.pm line 50.
BEGIN failed--compilation aborted at /tmp/.mount_Slic3rXe61ro/usr/bin/lib/site_perl/5.22.0/Slic3r.pm line 50.
Compilation failed in require at /tmp/.mount_Slic3rXe61ro/usr/bin/slic3r.pl line 16.
BEGIN failed--compilation aborted at /tmp/.mount_Slic3rXe61ro/usr/bin/slic3r.pl line 16.`

Installing `libglu1-mesa` helps.

_Is this a new feature request?_
No
#### STL/Config (.ZIP) where problem occurs
N/A

dlozinski, 2018-05-18:
> I am sorry, but it is your responsibility to have the OpenGL drivers
> installed. We should not bundle the OpenGL drivers with our AppImage.
>
> We can either document it or write a piece of code (maybe extend the start
> script?) so it will diagnose a missing OpenGL library, but I don't think we
> will manage it for this release.

Hi, I think diagnosing the missing library and showing proper error message
should be enough.

probonopd, 2019-09-21:
Something like this:

[image]

Imho this should be handled on the OS level, not by the application itself.

- https://gitlab.gnome.org/GNOME/nautilus/issues/1166
- https://bugs.kde.org/show_bug.cgi?id=409777
- https://bugzilla.xfce.org/show_bug.cgi?id=15714
- https://github.com/mate-desktop/caja/issues/1262

bubnikv, 2019-09-23:
>    - mate-desktop/caja#1262

We may try to load the opengl libraries on Linux explicitely and give some
reasonable error report if it is not possible, but it takes quite some
effort.

On Windows, we do exactly that to be able to load the software renderer if
OpenGL 2.0 is not avaiable from the system. We split the application to a
thin wrapper executable, and the whole slicer is then a dynamically loaded
library.

closed 2026-09-29 by prusaslicer-bot[bot]