prusa3d/PrusaSlicer #901: Slicing removes parts from model (1.39.2/1.40.0-alpha1)

ammulder, 2018-05-18:
### Version
Version 1.40.0-alpha1+ (also seen with 1.39.2)

### Operating system type + version
macOS 10.13.4

### Behavior
This model has rectangular holes in the base, and includes a vertical support in the middle of the hole (to prevent the top sagging, because a clip is expected to fit into this hole and much sagging will prevent the clip from fitting).  You are expected to remove the support manually after printing (e.g. with tweezers or needlenose pliers).
[image]

After slicing, the vertical support is gone:
[image]

#### STL/Config (.ZIP) where problem occurs
[model-and-config.zip](https://github.com/prusa3d/Slic3r/files/2017686/model-and-config.zip)

_(If for some reason there is a flaw in this particular model, there are numerous models using this same base that show the same issue.)_

bubnikv, 2018-05-18:
Aren't they just too thin to be printed reliably? Will they show up when
you enable "detect thin walls"?

ammulder, 2018-05-18:
They print fine with other slicers.  They're not meant to be super-strong, just enough to prevent the top of the hole from sagging.

They do appear when I enable "detect thin walls" (I hadn't been aware of that setting) -- is that the solution, then?

bubnikv, 2018-05-18:
> They print fine with other slicers. 

That does not mean that the other slicers are right.

You are asking the slic3r to do impossible: To print a thin wall with a perimeter (two lines in parallel) with a 0.4mm wide nozzle I guess. A reasonable width would be 0.8mm. The Slic3r knows that it will not fit two 0.4mm wide lines into your thin object, therefore it gives up.

Slic3r even gives you a hint on the recommended thin wall width. Please note how the hint changes if you enable / disable "detect thin wall".

[image]

> They do appear when I enable "detect thin walls" (I hadn't been aware of that setting) -- is that the solution, then?

or increase their width.

ammulder, 2018-05-19:
I don't know what to say.

These are commercial models with "Dragonlock" bases.  They are in wide usage by the original company and many others who have licensed the base and connection scheme.  (I am not an employee of that company, so I have no ability to alter the official base used by all these companies.)  They've been printed on dozens of printers.  I've personally printed models with this base on MakerBot, Printrbot, Flashforge, and Monoprice printers before I got Mk3.  This is the first time the model has printed for me without that small support piece.  You can see a larger list of printers that people use for these models at http://fatdragongames.proboards.com/board/24/3d-printers (and those are only the ones where people have mentioned on the forums that they use it).

So you may be correct that the model is technically using bad dimensions for the small support there.  (I'm not sure if that's true, but I'm willing to take your word for it.)

However, to be practical, the model is in wide use and (as far as I'm aware) no other vendor's printers print it without that small support.  It would seem best if Prusa printers printed the part too...  Just so people like me aren't like "why did my new high-end Prusa printer not print this correctly?  All my older/lesser printers printed it just fine!"  :)

FWIW, it would be fine with me if it was printed with one single stroke.  (You mentioned two lines in parallel, but it really only needs to be one line for that piece.  Again, it's not supposed to look good or be sturdy -- it's only supposed to help stop the top from sagging until you rip it off with a needlenose.)  Is there some way slic3r can detect that this piece is too narrow for two parallel lines and just do one line instead?  But I would still want the regular external faces of the model to print with a 2-line perimeter.

Itox001, 2018-05-20:
You could also make your perimeter line width smaller, the thinner lines may actually fit there.

hisashime, 2018-05-22:
I think there is problem with slicing in this slic3r Prusa. When I use the original ver of Slic3r my parts preview is complete but slic3r Prusa seems to removes things. I will have to open the stl in Slic3r original then saves as stl and overwrite the original and then load it again in Slic3r Prusa and the missing part is no longer removed.

bubnikv, 2018-05-22:
First, the object is not a valid STL, it contains multiple overlapping, possibly intersecting geometries. This is an "exploded" view of your model:

[image]

So Slic3r is actually doing a good job processing this geometry. You may have a better luck, if you process your geometry through some STL fixing service such as https://tools3d.azurewebsites.net/

The thickness of your missing columns is 0.446mm. That is a thickness of a single line. If you enable the "detect thin features", you will get the line printed. I believe Slic3r does what it should.

@hisashime 
> When I use the original ver of Slic3r my parts preview is complete

Which original Slic3r version?

hisashime, 2018-05-22:
I am using slic3r v1.2.9 from slic3r.org and slic3r Prusa. My comment is not refering to this post's stl but my own stl which have similar issue. I slice using different slicer upon experiencing problem with Slic3r Prusa and discover there could be issue and would like to report it here.

bubnikv, 2018-05-22:
@hisashime Slic3r 1.2.9 is extremely outdated. I wonder how the current "dev" Slic3r handles it. I bet it will handle the model the same way as the Prusa3D slic3r. 
http://slic3r.org/download

ammulder, 2018-05-22:
Regarding the specific model in this post: it is also a third-party model, and I am not surprised that the geometry is not optimal.  (As an aside, the geometry elsewhere in this model printed better with 1.40.0 alpha than with 1.39.2, so thank you for that!)  But the base containing the support in question is a complete part in your exploded view -- I am assuming you're not saying that slic3r omits the small support in the clip hole because the geometry elsewhere in the model is messy.

If you'd like to review some of the "official" and presumably geometrically-correct models using this base, you can get them here:

https://www.drivethrurpg.com/product/197128/DRAGONLOCK-Ultimate-Free-Sample-Set

They are commercial models, so I can't just attach them to this issue, but the set above is available for free as a sample and includes two models using the base as well as the clip that's meant to fit into the hole in question.

Again, if you believe slic3r is doing the right thing, I won't object if you close this issue.  I just want to be clear, though, that literally every other printer I've used prints this support piece by default.  So you would be taking a stand by producing a correct-but-inferior print compared to other slicers and printers.  I hope that is not the position you wish to take, but the decision is yours to make.  (Perhaps a compromise would be to enable "detect thin walls" by default?  Then slic3r can both be "correct" and produce the desired print by default?)

If you want a sense of the scale of the issue, searching DriveThruRPG site for "Dragonlock" (the name for the clip-together connection scheme) reports 98 sets of models using this clip hole and support for sale: https://www.drivethrurpg.com/browse.php?keywords=dragonlock and the company's last kickstarter had 1400 supporters: https://www.kickstarter.com/projects/1679800548/dragonlock-3-dangerous-lairs -- so I think this is an issue that can be expected to affect a decent number of your customers.

closed 2026-09-29 by prusaslicer-bot[bot]