prusa3d/PrusaSlicer #903: Support 'integrated graphics only' mode on macOS

aequitas, 2018-05-18:
I recently switched from a single GPU (integrated) macbook to a dual GPU (integrated and dedicated) and have noticed that using Slic3r on this laptop (or having it running in the background) significantly reduces battery life (from 5-6 to 2-3 hours) where this behaviour would not manifest on my previous laptop. This has to do with Slic3r using OpenGL which causes the macbook to switch to the dedicate GPU (high perf, see below) instead of the battery friendly integrated one.  

[image]

Since Slic3r is not a highly demanding 3D application (it performed completely fine on my older machine with lesser integrated video card) I think it could be safe to have Slic3r run in the integrated mode for these kind of macs.

There is an option in the `Info.plist` which allows for this (`NSSupportsAutomaticGraphicsSwitching` https://developer.apple.com/library/content/qa/qa1734/_index.html), but adding it does not seem sufficient. Slic3r or the OpenGL framework used by Slic3r must have some changes to support this mode (https://developer.apple.com/library/content/technotes/tn2229/_index.html).

My knowledge of the Slic3r(PE) codebase is limited but afaict the OpenGL parts are implemented in Perl. With the intention to reduce Perl code in Slic3rPE would it be worth for me investigating this change for the current codebase or could this change better be included in a potential future UI rewrite?

bubnikv, 2018-05-18:
For the next release (1.41) we already started to rewrite the OpenGL code into C++, see the opengl_to_cpp branch.

The OpenGL context is managed by wxWidgets 3.0 for us. We will switch to wxWidgets 3.1 once we get rid of the wxWidgets Perl binding.

aequitas, 2018-05-18:
Thanks for the quick response. I'll checkout that branch instead for investigating this, if I get somewhere with it you'll see a patch emerge.

aequitas, 2018-05-18:
I'm running into some issues getting the build environment setup. @bubnikv where can I find the latest build instructions? Or does the `opengl_to_cpp` branch require additional steps for build not documented yet?

Afaict the only change (besides the `Info.plist` mentioned above) needed to be made to the code is to add the `NSOpenGLPFAAllowOfflineRenderers` attribute to `NSOpenGLPixelFormat`, which I think should happen somewhere here: https://github.com/wxWidgets/wxWidgets/blob/v3.0.4/src/osx/cocoa/glcanvas.mm#L81 
I haven't quiet figured out how this translates up the stack into the Slic3rPE codebase. 

See also: http://supermegaultragroovy.com/2016/12/10/auto-graphics-switching/

rtyr, 2021-01-20:
I believe we can close this issue now, since it is related to very old PS versions.

closed 2021-01-20 by rtyr