prusa3d/PrusaSlicer #905: No platter view

Cretemonkey, 2018-05-19:
### Version
_Version of Slic3r 1.39.2

### Operating system type + version
Windows 10

### Behavior
After using Slic3r Prusa Edition a lot over the last 3 weeks I opened it today and there is no platter view. It will import the STL file but it's just a white square. This was after a Win 10 update.

I did try to reinstall the software
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_
[image]

matt-3, 2018-05-28:
Same issue at the same time, on 21 May, after Win10 update. This renders impossible to use Slic3r anymore. I have tried ununstalling and reinstalling version 1.39.1, but that did not help.

bubnikv, 2018-05-29:
I am sorry to hear that. What graphics card is on your computer? We recommend to try to update the graphics driver from the manufacturer.

bubnikv, 2018-05-31:
> This is a GeForce GT 540M on a laptop. There has not been any driver
> release for a long time and the latest one is installed. I have seen no
> issue with software other than Silc3r Prusa.

Microsoft updates often mess up the graphics drivers. You may try to
reinstall the old graphics driver.

Many 3D graphics applications on Windows use the Direct3D application
interface, so it may not be obvious that the OpenGL interface is broken.

monoverse, 2018-06-01:
Same problem here with Slic3rPE-1.40.0-beta.linux64-full-201805231653.AppImage. Both plater preview and layer view are not working anymore. My laptop has Intel onboard graphics:

`$ glxinfo`
`...`
`Vendor: Intel Open Source Technology Center (0x8086)`
`Device: Mesa DRI Intel(R) Sandybridge Mobile  (0x126)`
`OpenGL version string: 3.0 Mesa 17.3.3`
`...`

`lspci -k`
`...`
`00:02.0 VGA compatible controller: Intel Corporation 2nd Generation Core Processor Family Integrated Graphics Controller (rev 09)`
`Subsystem: Lenovo 2nd Generation Core Processor Family Integrated Graphics Controller`
`Kernel driver in use: i915`
`Kernel modules: i915`
`...`

I've tested a number of slic3r versions and found 1.38.3 the last one to be working. The [changelog of 1.38.4](https://github.com/prusa3d/Slic3r/releases/tag/version_1.38.4) states under section improvements which might be related to the problem:

> Implemented workaround for buggy Intel HD Graphics OpenGL drivers: Fall back to OpenGL 1.1 and disable multi sample anti aliasing if the new "use_legacy_opengl" preferences checkbox is enabled #233 #268 #619.

bubnikv, 2018-06-04:
@monoverse Instead of the AppImage, you can just unpack the zip binary distribution we provide for Linux, and you can add the glFlush and glFinalize where they were in the 1.38.3. The glFlush and glFinalize were used inside the perl modlues, so you can add them without recompiling Slic3r. I would be very thankful, if you could find a combination, which works on your system. Maybe it would help to @Cretemonkey as well, and we may make a configuration switch to enable it, if it helps.

bubnikv, 2018-06-07:
> @bubnikv <https://github.com/bubnikv> Could you please be a bit more
> specific about the glFlush and glFinish entities you mentioned. Are these
> files? Where do I find them. Where do I need to copy them? Thx for your
> help.

This was the change I made to fix the issue on most of the Intel HD
Graphics computers:

https://github.com/prusa3d/Slic3r/commit/acce8dbff6d9959b6ec879a4989c55ae11fa1deb#diff-8cf21c60fb92537ecfc7bb0f527027ab

You may try to revert the changes and test if it helps on your system. If
it does, then we may implement some kind of switch into the preferences
dialog.

monoverse, 2018-06-07:
> This was the change I made to fix the issue on most of the Intel HD
> Graphics computers:
> 
> acce8db#diff-8cf21c60fb92537ecfc7bb0f527027ab

@bubnikv I'am not quite sure why you are referring to this specific commit, because the original regression was introduced with version 1.38.4 since 1.38.3 is the last version that works for me. So, I would suppose the issue has smth to do with 673e98b or 743fc9d. Please correct me if I'am wrong.

However, I tried toggling the new "use legacy OpenGL1.1 rendering" preferences switch in version 1.38.4. but to no avail. Preview tab does not show the model, and the Layers tab is completely empty.

bubnikv, 2018-06-07:
I believe the commit acce8db I referred to was introduced in  Slic3r Prusa Edition 1.39.1-beta1 first.
https://github.com/prusa3d/Slic3r/releases/tag/version_1.39.1-beta

In that case the alpha should still work.
https://github.com/prusa3d/Slic3r/releases/tag/version_1.39.1-alpha

monoverse, 2018-06-07:
> I believe the commit acce8db I referred to was introduced in Slic3r Prusa Edition 1.39.1-beta1 first.
> https://github.com/prusa3d/Slic3r/releases/tag/version_1.39.1-beta

yes, but the platter view is already broken with https://github.com/prusa3d/Slic3r/releases/tag/version_1.38.4 which is from 2017 Dec 14. So I still don't see the connection between acce8db and this issue.

matt-3, 2018-06-26:
I fixed the issue by reinstalling older VGA drivers provided by the laptop manufacturer (ACER) over the ones Windows had upgraded.
Still it is quite strange that Slic3r was the only software I am using having an issue with this, and only in platter view. Btw going back to 1.39.1 did not make any difference.

tracernz, 2018-06-27:
>Still it is quite strange that Slic3r was the only software I am using having an issue with this, and only in platter view. Btw going back to 1.39.1 did not make any difference. 

I suspect Slic3r has no fallback if the native GL driver fails. The default behaviour of Qt based applications, Chrome, etc. is to fall back to ANGLE and/or software rendering (MESA/Gallium) if the native GL driver fails, which is why many applications will still run with bad native GL drivers (common on Windows with lower grade graphics hardware sadly).

bubnikv, 2019-01-29:
Please retest with the current 1.42 alpha
https://github.com/prusa3d/slic3r/releases

We have the highest hopes that the issue is solved with the wxWidgets update and by getting rid of the Perl (Slic3r 1.42 is now native only).

bubnikv, 2019-04-13:
No response, closing.

closed 2019-04-13 by bubnikv