prusa3d/PrusaSlicer #907: Octoprint

merlin2105, 2018-05-20:
The Octoprint IP address is not saved and must be re-entered every time Slic3r is restarted..

### Version
1.40.0-alpha1

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

stelgenhof, 2018-05-21:
This has been resolved in the master branch (See https://github.com/prusa3d/Slic3r/issues/869)

vojtechkral, 2018-05-21:
This is most likely a duplicate of #869 

(If not, let us know.)