prusa3d/PrusaSlicer #909: [Request] Don't support short/tiny overhang threshold

sarf2k4, 2018-05-20:
One thing I noticed that never been implemented across slicers is "Don't support short/tiny overhang". What I have in mind to this proposal is not to add support on short overhang. 

Take 3dbenchy.stl for example and enable default support at 45' angle with support on everywhere. Notice the support are also generated at the hull's rail and the roof's extension as well as the chimney part shown in the picture. Those supported parts are actually angled at about 0-5' and there are area's we need support at about 30' angle but we don't want to generate on these tiny overhangs.
[image]

My proposal is, add a new option where the support won't be generated lesser than the threshold in mm (all direction compared to the previous layer) and there will be support if there's more than the threshold. Similar to the angle threshold but for tiny overhangs.

This also will save gcode processing time, materials used and the time taken to remove the support instead of covering the whole model with supports where these tiny overhangs can be printed perfectly without supports to begin with.

May sound so simple but what if this feature really gets implemented and we have better control over supports?

Itox001, 2018-05-20:
Sounds good, I've also had models where a whole support tower is generated just to support a figures eyelids, which would print fine otherwise. Problem is, for example in this model, the towers generated were too thin and they would obviously fail, and since those eyelids were very small 0 degree overhangs, no overhang threshold would get rid of them, making the model unsupportable with slic3r's supports.

sarf2k4, 2022-02-21:
I wonder if this is feature is still in development? I know this is one of the hardest feature to implement as there are algorithm that the slicer must calculate and find out the areas first, I wonder if the following idea could be implemented?

Use the existing auto support enforcer, then use netfabb stl repair algorithm only on those faces to somewhat find out the area coverage size. Perhaps might work, I don't know

I am attaching a perfect example of an object that one would wish this feature would be implemented.
[torus-knot-dragon-and-pearl.zip](https://github.com/prusa3d/PrusaSlicer/files/8111790/torus-knot-dragon-and-pearl.zip)

closed 2026-09-29 by prusaslicer-bot[bot]