prusa3d/PrusaSlicer #910: [Request] Change order of settings in right panel (Plater)

stelgenhof, 2018-05-21:
### Version
v1.40.0-alpha1
git rev-parse HEAD: deabeaaa7fc48691c9b778e11a725f7bd93a0458

### Operating system type + version
Arch Linux 4.16.9-1-ARCH

### Behavior
Currently the settings (combo boxes) on the right panel of the Plater window ("Print settings", "Filament" and "Printer") are in an order that make the user to do unnecessary mouse travel.

"Print settings" and "Filament" have a dependency on the selected printer, so if no printer (or a different printer) is selected, you need first go to the third combo box and then move up to choose the desired print settings and/or filament.

It is more logic that the "Printer" combobox is on top, followed by "Filament" and then "Print settings". This reduced the mouse travel of the user, and therefore more fluent UI/UX. 

_Is this a new feature request?_
Yes

Panayiotis-git, 2018-05-21:
Does this mean that the order of the tabs will also be changed?

bubnikv, 2018-05-21:
We are planning a bigger overhaul of the UI in the next development iteration, we are currently discussing the design changes.

There will certainly be improvements of the 3D scene manipulators, and we may do bigger changes of the application layout as well, for example moving some buttons as overlays into the 3D scene, making the parameter tabs optional (maybe we will let them float into a separate window?).

stelgenhof, 2018-05-21:
@bubnikv Thanks for the update! I'll refrain from submitting UI requests then :)

lf-, 2019-04-06:
Since this is still a bug in 1.42-beta, it is filed again as #2079.

bubnikv, 2019-04-13:
It was decided by our boss to keep is at it is. Closing.

closed 2019-04-13 by bubnikv

tkapler, 2019-10-12:
I'm +1 for this, this is something what i hate from the beggining of my work with prusa slicer and it still makes me mad, you should persuade your boss to change his mind. This is not about boss desicion, it is about what is right from the UX logic.

And from User eXperience:
1. first you select the printer/nozzle (if you have only one, than you do not need this settings at all.
2. secondly you select the material, this is something you almost always know before you start your work with slicer
3. third you are playing with the layer height
4. you are playing with the other settings. 
And this is my biggest issue - changing layer height turns off supports if they were selected, changes infill to default ... I HATE when i set up everything and then i realise, that the time for printing is too high, so i need to change layer hight, but then i have to go and again set everything.