prusa3d/PrusaSlicer #911: Changing nozzle diameter deletes all print profiles

Fargle2nd, 2018-05-21:
### Version
 1.39.2+win64

### Operating system type + version
MSWIN32

### Behavior
Changing the nozzle diameter under "printer setting" to 0.5mm deletes all the printer settings in the drop down menu on the main page, just giving one option, typically titled "unknown" or "untitled", no other printer options are available in the drop down menu.

stelgenhof, 2018-05-26:
Curious about this issue, I tried myself as well. For me it worked without any issue, in that I can change the nozzle diameter, save it and still all printer presets are retained. 

I'm using v1.40.0-beta on Linux. Could you give v1.40.0-beta a try?

bubnikv, 2018-08-02:
Old version's bug reported, no response. Closing.

closed 2018-08-02 by bubnikv