prusa3d/PrusaSlicer #913: Automatic mesh bed leveling causes oozed filament deposits to be placed on the bed

emreerokyar, 2018-05-21:
### Version
1.40.0-alpha1+

### Operating system type + version
macOS High Sierra 10.13.4

### Behavior
Slic3r PE performs automatic mesh bed leveling before starting a print as it's added by default into "Printer Settings > Custom G-code > Start G-code" with this line;

`G80 ; mesh bed leveling`

However, this line comes after;

```
M104 S[first_layer_temperature] ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for extruder temp
```

This causes the filament (the original, Prusa silver PLA that comes in the box) to start oozing before the print starts. When performing the mesh bed leveling, the nozzle comes very close to the bed at 9 specific coordinates for measurement, and since there is usually some leaked filament hanging out, it usually gets deposited on the bed, within the print area.

Although not much of a problem and only a small nuisance, those deposits usually catch the nozzle while printing the model, or stick to the model itself.

One possible workaround could be to retract the filament before performing mesh bed leveling, then priming this extra amount in the prime area.

* _Steps needed to reproduce the problem_
After starting any print job on the MK3, I can reproduce this issue while automatic mesh bed leveling is being done with the Prusa Silver PLA at all times.

* _Expected Results_
I clean the bed before starting a print, and expect it to stay clean until the print starts, which is not the case.

* _Actual Results_
Leftover filament deposits are placed on the bed before the actual print starts.

_Is this a new feature request?_
Yes. I believe the default "Start G-code" could include a retraction before this line;

`G80 ; mesh bed leveling`

...and then prime the nozzle as usual before starting the print.

tracernz, 2018-05-21:
So the proposal is either:
- Do mesh bed levelling before heating, or
- Retract the extruder before doing mesh bed levelling (after heating)?

I think either of those would be an improvement (and I often manually preheat, and retract, before starting the print).

On my Prusa I'd still preheat/retract manually before starting the print though, because the PINDA performance drifts a bit with temperature variation unfortunately. I believe it does have a method to calibrate the temperature variation but I haven't spent the time to try it.

stelgenhof, 2018-05-21:
You can update the start and end gcode sections of your printer to include a retraction. In my printer profile I have the following:

Start G-Code:
```
...
G28 W ; home all without mesh bed level
G1  E-2 F100 ; Retract 2mm at a speed of 100mm/s to prevent oozing when mesh bed leveling
G80 ; mesh bed leveling
...
```
End G-Code:
```
...
G1 E-2 F100 ; Retract 2mm at a speed of 100mm/s to release internal pressure preventing oozing at next print.
...
```

I don't feel this is an issue with Slic3r though, as you can easily mitigate it as shown above :)

emreerokyar, 2018-05-22:
@tracernz, I think performing automatic mesh bed leveling before hitting the actual print temperatures for both the bed and the nozzle could create issues with the PINDA probe and measurements. So I would not prefer your first option.

@stelgenhof, yes we can mitigate the issue as you showed above. And since it works, why not make it the default for all Slic3r PE users as I proposed - one less thing to worry about?

cnlson, 2018-05-23:
@emreerokyar I was similarly annoyed with this but have solved in the following manner:

_First I added a line to set the temp to 140 which should not ooze for any filament_
M104 S140 

M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
_then i moved the first layer print temp to after the bed is at temp_
M104 S[first_layer_temperature] ; set extruder temp
M109 S[first_layer_temperature] ; wait for extruder temp

This works 100% of the time for printing and mostly eliminates the oozing since the print head is fully headed for a very short time before 9 point levelling.

bubnikv, 2018-06-04:
So you guys could fix the issue by editing the start G-code and we may close this issue?

stelgenhof, 2018-06-04:
For me personally this isn’t much of an issue as you can easily add some custom gcodes. Some prefer the ooze prevention to be included by default in the Slic3r settings.

Perhaps some users wouldn’t want this as a default so I feel adding this to your own settings is perfectly fine.

gdombiak, 2018-06-05:
In the spirit of simplifying user experience, I think that it would be nice if Slic3r could solve this out of the box. Depending on each user to play with gcode to take care of this annoyance is not the best UX.

Petclaud, 2018-06-05:
Yes agree, suggest to include it in Slic3r print profiles by default. 

In addition; I am not a GCODE expert, but I guess that with: 

`G1  E-2 F100 ; Retract 2mm at a speed of 100mm/s to prevent oozing when mesh bed leveling
`
and with

`G1 E-2 F100 ; Retract 2mm at a speed of 100mm/s to release internal pressure preventing oozing at next print. `

the speed is 100mm/minute in stead of 100mm/second and hence is a bit slow...?  Maybe 2000mm/minute is a better speed, so:

G1  E-2 F2000...?

bubnikv, 2018-06-06:
> @Petclaud <https://github.com/Petclaud> Thanks for catching this :) It's
> only a typo in the comment. (should be 100mm/minute).
>
> If you like it to be faster, you can adjust accordingly. I prefer this
> feedrate of 100mm/minute and don't mind it being a bit slow :)

> G1 E-2 F100 ; Retract 2mm at a speed of 100mm/s to prevent oozing when
mesh bed leveling

This is a dangerous proposition, which may lead to extruder clogging. The
problem is, at the start of the print you don't know the position of the
filament in the heat break. If you retract too far, a clog may form with
PLA.

Panayiotis-git, 2018-06-06:
On the other hand, it is absolutely safe to use it at the end g-code. In fact, I retract the filament 4mm at the printing end without any problem.

bubnikv, 2018-06-06:
We need the firmware to "know" the position of the filament parked in the
nozzle when idle. For example, when loading the filament, would we like to
retract the filament by 4mm after the filament loading finishes?

The whole issue is more complex than you guys think. For our support it is
far easier to explain a bit of oozing at the time of bed leveling than to
explain the clogged nozzle due to filament expansion at a bad moment at a
bad position in the heat break.

bubnikv, 2018-06-06:
> @bubnikv <https://github.com/bubnikv> Thanks for the information: I
> wasn't aware that it indeed can cause potential clogs. Up until now I
> haven't had any issues yet, though.
>
> I understand this issue can be quite complex. From my side, this isn't an
> urgent issue, so if you and your team have another idea/solution, I am
> happy to follow that.
>
> Cheers! Sacha

We were discussing internally to do a bed leveling at the 1st point only,
then to do the extruder priming, then to retract, then to do the full 9
point bed leveling, and then finally to deretract and extrude a short 2nd
priming line. This should solve the leakage.

Is anybody willing to help to prototype that?

stelgenhof, 2018-06-06:
@bubnikv Sure, count me in!

bubnikv, 2018-06-06:
@stelgenhof Thanks, it sounds like one will need to do at least couple of iterations before it works as expected.

fiveangle, 2018-06-06:
@bubnikv - since you are refining this process, can I suggest that you turn-on the cooling fan and include a fast-move and nozzle-drop to the right of the bed for the purge lines ?  I have found it will prevent any stringing present from being dragged into the print area which can cause issues.  I drop the nozzle to close to the bed surface from purge line print height, and move to the right at max acceleration/speed rate to within a few CM of right bed edge and it has worked excellently to eliminate the string from the heavy purge.  Please let me know if this is not clear.

Thanks !

-=dave

Petclaud, 2018-06-06:
Good that is being looked at and improved. Just another thought/suggestion; also during pre-heating oozing occurs. I guess this cannot be prevented. As as suggestion to ooze preventing during mesh bed leveling, maybe the following also helps: preheat the nozzle e.g. 5 degrees higher than the fist layer printing temperature and once the print command is given and this +5 degrees higher is reached (and oozing occurs and can be removed), the temperature is lowered again to the actual 1 st layer temperature and only once that  temperature is reached, the mesh bed leveling takes place. See also here: [https://github.com/prusa3d/Prusa-Firmware/issues/699](url)

bubnikv, 2018-08-02:
@jindrichbenes will try the solution with lowering the temperature a bit before bed leveing. Let's see how reliable it will work.

ldpgh, 2018-11-25:
I do have the same issue. My setup is:
- Slicr3r Prusa Edition Version 1.41.1-rc+win64
- Original Prusa i3 MK3 kit @ Oct 2018
- White Renkforce 1.5mm PLA using default PRUSA PLA profiles.

I have modified the "Start G-code" sequence of the all PRUSA PLA-profiles by adding preheating to 170degC ... see below for LDA. The temperature is low enough to prevent ouzing during calibration.

In addition
- the number of Skirt loops has been set to 3
- Skirt height has been reduced to 1 layer. When the issue appears the 2nd skirt layer makes it worse while traveling between object and Skirt with material attached outside to the nozzle. Btw. I use the skirt to measure 1st layer height with the caliper.

```
....
M115 U3.3.1 ; tell printer latest fw version
M83  ; extruder relative mode
M104 S170 ; set extruder temp           LDA 1a...line modified...temp reduced to prevent oozing
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S170 ; wait for extruder temp      LDA 1b...line modified...temp reduced to prevent oozing
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling
G1 Y-3.0 F1000.0 ; go outside print area
M104 S[first_layer_temperature] ; set extruder temp        LDA 1c...line added...bring-up extruder temp finally
M109 S[first_layer_temperature] ; wait for extruder temp   LDA 1d...line added...bring-up extruder temp finally
G92 E0.0
G1 X60.0 E9.0  F1000.0 ; intro line
G1 X100.0 E12.5  F1000.0 ; intro line
G92 E0.0
M221 S{if layer_height==0.05}100{else}95{endif}
...
```

rennne, 2019-01-31:
It would also help if the bed leveling would end the left side back corner instead of the right side corner.

With that the oozing/stringing would not go diagonally (through the center) over the biuld platte but on the left side where most of the time no print object will be.

This would be especially interesting for filaments  that ooze or string even with retraction (PETG) or tend to clog when being retracted (FLEX).

StephenRC, 2019-01-31:
You could just add a G28 X after the leveling gcode in your startup
gcode section.

On Thu, 31 Jan 2019 09:43:59 -0800, rennne <notifications@github.com>
had a flock of green cheek conures squawk out:

> It would also help if the bed leveling would end the left side back corner instead of the right side corner.
>
> With that the oozing/stringing would not go diagonally (through the center) over the [...]

-- 

Stephen

. . . . File not found. Should I fake it? Y/N

---
This email has been checked for viruses by AVG.
https://www.avg.com

rennne, 2019-01-31:
I tried it and unfortunately G28 X after the leveling gcode do not work.

It moves diagonally across the buildplatte and does the x-homing afterwards.

StephenRC, 2019-02-01:
Then use a G1 X0. If you use relative positioning, don't forget to
turn it off for the move.

On Thu, 31 Jan 2019 13:18:45 -0800, rennne <notifications@github.com>
had a flock of green cheek conures squawk out:

>I tried it and unfortunately G28 X after the leveling gcode do not work.
>
>It moves diagonally across the buildplatte and does the x-homing afterwards.

-- 

Stephen

. . . . File not found. Should I fake it? Y/N

---
This email has been checked for viruses by AVG.
https://www.avg.com

StephenRC, 2019-02-02:
https://reprap.org/wiki/G-code#G80:_Cancel_Canned_Cycle_.28CNC_specific.29
G80: Cancel Canned Cycle (CNC specific)

Also is not a valid code in RRF.

On Sat, 02 Feb 2019 02:22:42 -0800, rennne <notifications@github.com>
had a flock of green cheek conures squawk out:

>G1 X0 does not work because it fixed in the firmware
>Quote form another thread:
>> G80 does mesh bed levelling and then homes via G28.
>>[https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/Marlin_main.cpp#L4574](https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/Marlin_main.cpp#L4574)
>>imho it would be sensible to separate that, especially since the default start g-code in Slic3rPE goes to the start position for the purge line immediately afterwards anyway.

-- 

Stephen

. . . . File not found. Should I fake it? Y/N

---
This email has been checked for viruses by AVG.
https://www.avg.com

tsprhdm, 2019-07-02:
We currently use 7 i3MK3 in our university lab to teach 3d printing. Most of our students "just need something printed", and that's perfectly okay.  We do show them how to change filament, and they do that very regularly. Students will run the prusas at their own convenience, since we're not available 24/7.
Overall, this works preety darn good with the prusas.

Since we encourage our students to always get the newest version of prusaslic3r to prepare their files, the oozing during mesh leveling and filament strings/blobs on the nozzle at the start of first layer has become an issue. Even more so, since we have 2 printers with .25 nozzles, and we like to run 7x7 mesh leveling on those.

A few suggestions / considerations:

- Since people will download a new version / config updates on their personal computers all the time, the change of the start code needs to be rolled out through prusa. 

- Even on our lab PC's, the printer config updates are stored into the windows user profile folder. Since our students log on using the active directory using their uni credentials, we can#t provied a lab override. The fix needs to be in the prusa software / config update rolled out by prusa.

Have you considered doing mesh bed leveling at a fixed bed and nozzle temperature? E.G. always level at 50°C Bed, 170°C Nozzle?

...
M190 S50 ; wait for bed to come up to calibration temp.
M109 S170 ; wait for nozzle to come up to calibration_temp.
M80 ; execute mesh bed leveling
M140 S[first_layer_temperature] ; set nozzle temperature
M190 S[first_layer_bed_temperature] ; wait for bed temp
M190 S[first_layer_temperature] ; make sure nozzle is at correct temperature
...

Like so, even the temperature drift of the p.i.n.d.a. is not an issue any more, since it's always being probed at the same temp, and there should not be an oozing issue with most of the materials at something like 150 to 180 °C

tsprhdm, 2019-07-02:
I just tried, works a treat.
Whether to keep the bed at fixed 50 °C or mesh level at first-layer-temp needs some discussion, though, since the mechanical design calls for temperature-induced warping of the bed.

Final trial code:

M190 S50 ; wait for bed to come up to calibration temp
M109 S170 ; wait for nozzle to come up to calibration_temp

G28 W ; home all without mesh bed level
G80 ; execute mesh bed leveling
G1 X0 Y0 Z50 F1000.0 ; present nozzle for clean check and heatup

M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; make sure nozzle is at correct temperature

G1 X10 Y-3.0 Z0.20 F1000.0 ; go outside print area

tsprhdm, 2019-07-02:
Just found out, this is the wrong tree.

please bark at:

https://github.com/prusa3d/PrusaSlicer-settings/issues/10

rtyr, 2020-03-24:
Similar issues are opened here https://github.com/prusa3d/PrusaSlicer-settings/issues/10 and here https://github.com/prusa3d/PrusaSlicer-settings/pull/19 (and somewhere in the FW repo). It is more related to the settings and firmware. "Anti-oozing" start gcode is currently used on the Prusa MINI (because it oozes more), we will observe feedback from the customers. There are quite different opinions about this, because it is adding significant amount of time before the print. We might need some support in the firmware to reduce that time (cooling down from preheated state) or make this "feature" optional.

closed 2020-03-24 by rtyr