prusa3d/PrusaSlicer #915: Support generated even if support on build only activated

kakou-fr, 2018-05-23:
### Version
slic3r-1.40.0-alpha1+

### Operating system type + version
OS X 10.23.4

### Behavior
Slic3r generate strange support in this hole, even if support on build only activated.

Expected behaviour : support only on build plate

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
[image]
[image]
[image]

3Delights, 2018-05-27:
I think the supports system is in need of a major rewrite!

chadwixk, 2018-06-05:
Same issue on windows version 1.40.0-b. But this was also an issue on 1.38.4
[image]
[image]

aaronevers, 2018-06-14:
I see this too.     
[image]
[thistle.zip](https://github.com/prusa3d/Slic3r/files/2104331/thistle.zip)

gromain, 2018-09-25:
Same problem seen here as well, when slicing a simple sphere with a cut on the bottom (as an example).

[image]
Support on build plate only is selected, and I see two problems:
- Useless support is generated (See the part on the right, goes at an angle then stop)
- Support material is generated above sliced part inside the sphere

[support_test_0.2mm_PLA_MK3.amf.zip](https://github.com/prusa3d/Slic3r/files/2415006/support_test_0.2mm_PLA_MK3.amf.zip)

Linux with Slic3r PE 1.41.0+linux64 from AppImage

bubnikv, 2021-03-18:
With the current master

[image]
[image]

however some of the support towers trimmed to not land on object's top surface may not be stable.

bubnikv, 2021-03-19:
Fixed with 9f09f03228478aa77549efec845717b06e46cbc3

closed 2021-03-19 by bubnikv