prusa3d/PrusaSlicer #919: Plater preview keeps displaying only 1 layer

MiroPiencak, 2018-05-24:
### Version
slic3r-1.4.0-beta

### Operating system type + version
MacOS 10.13.4

### Behavior
* Slic3r displays only 1 layer in the plater preview screen even if theis option has been deactivated.
* _STR_
  * load a model
  * slice it
  * open the plater preview screen
  * check "1 Layer" option on the right
  * uncheck the "1 Layer" option
* _Expected results_
  * when the option is checked: only single layer is visible
  * after deactivating the option: all layers are visible again
* _Actual Results_
  * when selected, only 1 layer is visible
  * after unchecking the 1 Layer option, only one layer keeps visible

[image]

FidelCapo, 2018-05-24:
Did you try to move the left slider back to the first layer and right to the last layer? The error did not occur to me.

MiroPiencak, 2018-05-24:
@FidelCapo You are right, when the left slider is moved, more layers become visible. 
In that case it is not a bug.

closed 2018-05-24 by MiroPiencak