prusa3d/PrusaSlicer #920: Unable to save settings

Vance1122, 2018-05-24:
### Version
_Version of Slic3r used goes here_
Slic3r PE 1.40.0 beta Win64
_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10 with latest updates.
### Behavior
* _Describe the problem_When I change the infill style and want to save it, it won't do it
* _Steps needed to reproduce the problem_change a setting and try to save it
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_to save the setting
* _Actual Results_does not save the setting
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2018-05-24:
Aren't you trying to save into a system preset?

Please note, that the Prusa's presets are now read only, and you have to
rename your preset to be able to save it.

Vance1122, 2018-05-24:
Well, that's silly.

bubnikv, 2018-05-24:
> Well, that's silly.

So is this issue solved for you? Do you have any comment on how to make the feature more obvious?

bubnikv, 2018-06-04:
It seems to have been a misunderstanding of the new system presets concept. Closing.

closed 2018-06-04 by bubnikv